open-mmlab/OpenPCDet #1005: NaN or Inf found in input tensor in NUSCENES

JianyyuWang, 2022-06-19:
2022-06-19 11:16:02,958   INFO  **********************Start logging**********************
2022-06-19 11:16:02,958   INFO  CUDA_VISIBLE_DEVICES=0, 1
2022-06-19 11:16:02,958   INFO  cfg_file         cfgs/nuscenes_models/cbgs_pv_rcnn_multihead_no_velo.yaml
2022-06-19 11:16:02,958   INFO  batch_size       4
2022-06-19 11:16:02,958   INFO  epochs           20
2022-06-19 11:16:02,958   INFO  workers          4
2022-06-19 11:16:02,958   INFO  extra_tag        default
2022-06-19 11:16:02,958   INFO  ckpt             None
2022-06-19 11:16:02,958   INFO  pretrained_model None
2022-06-19 11:16:02,958   INFO  launcher         none
2022-06-19 11:16:02,958   INFO  tcp_port         18888
2022-06-19 11:16:02,958   INFO  sync_bn          False
2022-06-19 11:16:02,958   INFO  fix_random_seed  False
2022-06-19 11:16:02,958   INFO  ckpt_save_interval 1
2022-06-19 11:16:02,958   INFO  local_rank       0
2022-06-19 11:16:02,958   INFO  max_ckpt_save_num 30
2022-06-19 11:16:02,958   INFO  merge_all_iters_to_one_epoch False
2022-06-19 11:16:02,958   INFO  set_cfgs         None
2022-06-19 11:16:02,958   INFO  max_waiting_mins 0
2022-06-19 11:16:02,958   INFO  start_epoch      0
2022-06-19 11:16:02,958   INFO  num_epochs_to_eval 0
2022-06-19 11:16:02,958   INFO  save_to_file     False
2022-06-19 11:16:02,958   INFO  cfg.ROOT_DIR: /root/workspace/OpenPCDet
2022-06-19 11:16:02,958   INFO  cfg.LOCAL_RANK: 0
2022-06-19 11:16:02,958   INFO  cfg.CLASS_NAMES: ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone']
2022-06-19 11:16:02,958   INFO  
cfg.DATA_CONFIG = edict()
2022-06-19 11:16:02,958   INFO  cfg.DATA_CONFIG.DATASET: NuScenesDataset
2022-06-19 11:16:02,958   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/nuscenes
2022-06-19 11:16:02,958   INFO  cfg.DATA_CONFIG.VERSION: v1.0-trainval
2022-06-19 11:16:02,958   INFO  cfg.DATA_CONFIG.MAX_SWEEPS: 10
2022-06-19 11:16:02,958   INFO  cfg.DATA_CONFIG.PRED_VELOCITY: False
2022-06-19 11:16:02,958   INFO  cfg.DATA_CONFIG.SET_NAN_VELOCITY_TO_ZEROS: True
2022-06-19 11:16:02,958   INFO  cfg.DATA_CONFIG.FILTER_MIN_POINTS_IN_GT: 1
2022-06-19 11:16:02,958   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2022-06-19 11:16:02,958   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2022-06-19 11:16:02,958   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2022-06-19 11:16:02,959   INFO  
cfg.DATA_CONFIG.INFO_PATH = edict()
2022-06-19 11:16:02,959   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['nuscenes_infos_10sweeps_train.pkl']
2022-06-19 11:16:02,959   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['nuscenes_infos_10sweeps_val.pkl']
2022-06-19 11:16:02,959   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [-51.2, -51.2, -5.0, 51.2, 51.2, 3.0]
2022-06-19 11:16:02,959   INFO  cfg.DATA_CONFIG.BALANCED_RESAMPLING: True
2022-06-19 11:16:02,959   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2022-06-19 11:16:02,959   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2022-06-19 11:16:02,959   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'DB_INFO_PATH': ['nuscenes_dbinfos_10sweeps_withvelo.pkl'], 'PREPARE': {'filter_by_min_points': ['car:5', 'truck:5', 'construction_vehicle:5', 'bus:5', 'trailer:5', 'barrier:5', 'motorcycle:5', 'bicycle:5', 'pedestrian:5', 'traffic_cone:5']}, 'SAMPLE_GROUPS': ['car:2', 'truck:3', 'construction_vehicle:7', 'bus:4', 'trailer:6', 'barrier:2', 'motorcycle:6', 'bicycle:6', 'pedestrian:2', 'traffic_cone:2'], 'NUM_POINT_FEATURES': 5, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': True}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x', 'y']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.3925, 0.3925]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2022-06-19 11:16:02,959   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2022-06-19 11:16:02,959   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2022-06-19 11:16:02,959   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp']
2022-06-19 11:16:02,959   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp']
2022-06-19 11:16:02,959   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': True}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.1, 0.1, 0.2], 'MAX_POINTS_PER_VOXEL': 10, 'MAX_NUMBER_OF_VOXELS': {'train': 60000, 'test': 60000}}]
2022-06-19 11:16:02,959   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/nuscenes_dataset_no_velocity.yaml
2022-06-19 11:16:02,959   INFO  
cfg.MODEL = edict()
2022-06-19 11:16:02,959   INFO  cfg.MODEL.NAME: PVRCNN
2022-06-19 11:16:02,959   INFO  
cfg.MODEL.VFE = edict()
2022-06-19 11:16:02,959   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2022-06-19 11:16:02,959   INFO  
cfg.MODEL.BACKBONE_3D = edict()
2022-06-19 11:16:02,959   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x
2022-06-19 11:16:02,959   INFO  
cfg.MODEL.MAP_TO_BEV = edict()
2022-06-19 11:16:02,959   INFO  cfg.MODEL.MAP_TO_BEV.NAME: HeightCompression
2022-06-19 11:16:02,959   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 256
2022-06-19 11:16:02,959   INFO  
cfg.MODEL.BACKBONE_2D = edict()
[... 724 lines cut ...]
  File "/root/workspace/OpenPCDet/tools/train.py", line 211, in <module>
    main()
  File "/root/workspace/OpenPCDet/tools/train.py", line 163, in main
    train_model(
  File "/root/workspace/OpenPCDet/tools/train_utils/train_utils.py", line 110, in train_model
    accumulated_iter = train_one_epoch(
  File "/root/workspace/OpenPCDet/tools/train_utils/train_utils.py", line 46, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/root/workspace/OpenPCDet/tools/../pcdet/models/__init__.py", line 43, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/root/anaconda3/envs/OpenPCDet/lib/python3.9/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/root/workspace/OpenPCDet/tools/../pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/root/anaconda3/envs/OpenPCDet/lib/python3.9/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/root/workspace/OpenPCDet/tools/../pcdet/models/roi_heads/pvrcnn_head_for_nuscenes_v1.py", line 238, in forward
  File "/root/workspace/OpenPCDet/tools/../pcdet/models/roi_heads/roi_head_template.py", line 135, in assign_targets
  File "/root/workspace/OpenPCDet/tools/../pcdet/models/roi_heads/target_assigner/proposal_target_layer.py", line 37, in forward
  File "/root/workspace/OpenPCDet/tools/../pcdet/models/roi_heads/target_assigner/proposal_target_layer.py", line 138, in sample_rois_for_rcnn
  File "/root/workspace/OpenPCDet/tools/../pcdet/models/roi_heads/target_assigner/proposal_target_layer.py", line 190, in subsample_rois
NotImplementedError

when i was training on the Nuscenes datasets， i found this bug， and dont know how to fix it.
I tried change the spconv 2.1.21 -> 2.1.1 2.0.2 ,it still has the follow problem. my pcdet 0.5.2 pytorch==1.8.0 cuda11.1

closed 2022-08-04 by github-actions[bot]

zbaishancha, 2023-06-07:
@JianyyuWang Hi, hello, I have the same problem as you, do you know the solution yet?

OuYaozhong, 2023-07-16:
Hi, @JianyyuWang @zbaishancha @sshaoshuai @jihanyang @chenshi3 
I met this problem too.

1. 
The original learning rate in the tools/cfgs/nuscenes_models/cbgs_voxel0075_res3d_centerpoint.yaml is 0.001

But i have modified to with 100 times smaller as 0.00001, it still occur the NaN problem.

The loss is NaN and then the tensorboard throw the info that "NaN of Inf found in the tensor" message.

I have tried and proved that this problem have no relation with if using multiple GPUs.

I can not understand why cause this problem and reduce the learning rate with 100x smaller still unable to fixed this.

2. 
Meanwhie, the program often hang by random, with the same message that "collective operations have faid with timeout", and with 100% GPU utilisation.

It seems that it often hang at the dist.all_together() operation.

Could anybody fix above two problem ?

chenshi3, 2023-07-17:
It appears that the loss becomes NaN as soon as the training procedure begins. You might need to try debugging the code on a single GPU and monitor the loss calculation.

OuYaozhong, 2023-07-17:
> It appears that the loss becomes NaN as soon as the training procedure begins. You might need to try debugging the code on a single GPU and monitor the loss calculation.

Hi, @chenshi3

For my case, it is not.

1. The codes I just pull down from github without modification, besides some bug fix to make it run in my machine and environment.
2. Actually, it is true that the NaN will occur with 50% possibility at the beginning. **But sometimes it will run normally to 75% of 1st epoch, and then turn to NaN loss.**

I am using the OpenPCDet for nuScences dataset and CenterPoint.

Furthermore, I have tried the original CenterPoint, which release by its author. It seems same problem will occur with the default setting.

My environment:
```
$ nvidia-smi
Mon Jul 17 11:47:10 2023       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.54.03              Driver Version: 535.54.03    CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 3090        Off | 00000000:67:00.0 Off |                  N/A |
| 61%   54C    P2             142W / 350W |  10617MiB / 24576MiB |    100%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   1  NVIDIA GeForce RTX 3090        Off | 00000000:68:00.0 Off |                  N/A |
| 79%   59C    P2             154W / 350W |  10006MiB / 24576MiB |    100%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      1644      G   /usr/lib/xorg/Xorg                            4MiB |
|    0   N/A  N/A    875026      C   ........./.conda/envs/pcdet/bin/python    10600MiB |
|    1   N/A  N/A      1644      G   /usr/lib/xorg/Xorg                           14MiB |
|    1   N/A  N/A    875027      C   ........./.conda/envs/pcdet/bin/python     9978MiB |
+---------------------------------------------------------------------------------------+
```

```
$ conda list | grep torch
ffmpeg                    4.3                  hf484d3e_0    pytorch
pytorch                   2.0.1           py3.11_cuda11.8_cudnn8.7.0_0    pytorch
pytorch-cuda              11.8                 h7e8668a_5    pytorch
pytorch-mutex             1.0                        cuda    pytorch
torchaudio                2.0.2               py311_cu118    pytorch
torchtriton               2.0.0                     py311    pytorch
torchvision               0.15.2              py311_cu118    pytorch
```

```
$ echo $CUDA_HOME
/usr/local/cuda-11.8/
```