open-mmlab/OpenPCDet #1006: NaN or Inf found in input tensor

JianyyuWang, 2022-06-19:
I find that if anchor_head_multi.py was used on nuscenes and RCNN is used, it will cause an error like NaN or Inf found in input tensor.
Does Someone know what casues that ?

OrangeSodahub, 2022-06-23:
Also occurred this, have you solved it ?

JianyyuWang, 2022-06-26:
> 

not yet, I tried to change spcnv2.X to spconv 1.X ,but failed

OrangeSodahub, 2022-07-18:
I've solved it. Make sure that label files are not empty!

JianyyuWang, 2022-08-05:
> 

I do not quite understand . the label files is the official, you mean that we need to clear away the empty label files?

JianyyuWang, 2022-08-05:
> I've solved it. Make sure that label files are not empty!

And I checked the label files. when error occured, the label files are not empty

OrangeSodahub, 2022-08-05:
@JianyyuWang  I used my own custom dataset. For my dataset, it exactly helped.

TaigaMoriguchi, 2022-09-09:
I was able to solve this by updating the learning rate.

closed 2022-10-24 by github-actions[bot]

OuYaozhong, 2023-07-12:
Same problem. Anybody solved it ? Just use the official nuScene and follow the steps of INSTALL.md.

SimonSongg, 2023-09-05:
Same issue, I've clean up the samples without object. 

> Same problem. Anybody solved it ? Just use the official nuScene and follow the steps of INSTALL.md. @JianyyuWang

Although I used my own dataset, I met this issue when using CenterPoint. Finally I found there were some annotation issue: the height of the bounding box of some targets were negative. And the 3-dim of the bounding box takes .log(). Python returns NaN when trying to take logarithm of a negative number.

I hope this helps.