open-mmlab/OpenPCDet #1007: how to open result.pkl generated by train.py

1uk4s, 2022-06-19:
Hi there,
I trained a model and wanted to open the result.pkl file generated by train.py.

`import torch`
`res = torch.load("/path.../result.pkl")`
`print(res)`

but i get the error:
`RuntimeError: Invalid magic number; corrupt file?`

Now I am not sure if something went wrong with the generation of the file or if I open it wrong.
My question is: how do you look in the results in the result.pkl file ?

thanks in advance
Lukas

udacityyy, 2022-06-20:
Hello, I found this repo that allows to visualize the test result : https://github.com/DeclK/3d-object-vis

aldipiroli, 2022-07-20:
You can have a look on how the evaluation script opens the `.pkl` files.
[Here](https://github.com/open-mmlab/OpenPCDet/blob/c233477a4a4bf2861a978794d6b9c047f64641f7/pcdet/datasets/waymo/waymo_eval.py#L226) is how to load it for the `waymo_eval.py` script.

closed 2022-09-03 by github-actions[bot]