open-mmlab/OpenPCDet #1011: launch.py: error: unrecognized arguments: --rdzv_endpoint=localhost:30032

HuangZhe885, 2022-06-22:
when I use muti-GPU ,it got this problem
and when I detele,--rdzv_endpoint=localhost:${PORT}  ,  it works with another problem

subprocess.CalledProcessError: Command '['/home/hz/miniconda3/envs/BASA/bin/python', '-u', 'train.py', '--local_rank=3', '--launcher', 'pytorch', '--cfg_file', 'cfgs/kitti_models/3dssd_basa.yaml']' returned non-zero exit status 1.

jihanyang, 2022-07-04:
It seems that you haven't present all error messages.

closed 2022-08-18 by github-actions[bot]

wsy2711, 2023-03-02:
> when I use muti-GPU ,it got this problem and when I detele,--rdzv_endpoint=localhost:${PORT} , it works with another problem
>
> subprocess.CalledProcessError: Command [...]

hello! Did you slove this problem? I met this problem today TT
你好！请问您解决这个问题了吗？我刚刚也遇到了这个问题（哭）