open-mmlab/OpenPCDet #1014: installation Error " fatal error: THC/THC.h: No such file or directory  #include <THC/THC.h> "

ammaryasirnaich, 2022-06-24:
After cloning and installing the framework I am getting the below error, help will be much appreciated!
```
OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/sampling.cpp:4:10: fatal error: THC/THC.h: No such file or directory
    4 | #include <THC/THC.h>
      |          ^~~~~~~~~~~
compilation terminated.
[7/13] c++ -MMD -MF /workspace/OpenPCDet/build/temp.linux-x86_64-cpython-38/pcdet/ops/pointnet2/pointnet2_stack/src/pointnet2_api.o.d -pthread -B /opt/conda/compiler_compat -Wno-unused-result -Wsign-compare -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /opt/conda/include -fPIC -O2 -isystem /opt/conda/include -fPIC -I/opt/conda/lib/python3.8/site-packages/torch/include -I/opt/conda/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/include/python3.8 -c -c /workspace/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/pointnet2_api.cpp -o /workspace/OpenPCDet/build/temp.linux-x86_64-cpython-38/pcdet/ops/pointnet2/pointnet2_stack/src/pointnet2_api.o -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1013"' -DTORCH_EXTENSION_NAME=pointnet2_stack_cuda -D_GLIBCXX_USE_CXX11_ABI=1 -std=c++14
[8/13] /usr/local/cuda/bin/nvcc  -I/opt/conda/lib/python3.8/site-packages/torch/include -I/opt/conda/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/include/python3.8 -c -c /workspace/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/ball_query_gpu.cu -o /workspace/OpenPCDet/build/temp.linux-x86_64-cpython-38/pcdet/ops/pointnet2/pointnet2_stack/src/ball_query_gpu.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1013"' -DTORCH_EXTENSION_NAME=pointnet2_stack_cuda -D_GLIBCXX_USE_CXX11_ABI=1 -gencode=arch=compute_52,code=sm_52 -gencode=arch=compute_60,code=sm_60 -gencode=arch=compute_61,code=sm_61 -gencode=arch=compute_70,code=sm_70 -gencode=arch=compute_75,code=sm_75 -gencode=arch=compute_80,code=sm_80 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
[9/13] /usr/local/cuda/bin/nvcc  -I/opt/conda/lib/python3.8/site-packages/torch/include -I/opt/conda/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/include/python3.8 -c -c /workspace/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/interpolate_gpu.cu -o /workspace/OpenPCDet/build/temp.linux-x86_64-cpython-38/pcdet/ops/pointnet2/pointnet2_stack/src/interpolate_gpu.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1013"' -DTORCH_EXTENSION_NAME=pointnet2_stack_cuda -D_GLIBCXX_USE_CXX11_ABI=1 -gencode=arch=compute_52,code=sm_52 -gencode=arch=compute_60,code=sm_60 -gencode=arch=compute_61,code=sm_61 -gencode=arch=compute_70,code=sm_70 -gencode=arch=compute_75,code=sm_75 -gencode=arch=compute_80,code=sm_80 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
[10/13] /usr/local/cuda/bin/nvcc  -I/opt/conda/lib/python3.8/site-packages/torch/include -I/opt/conda/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/include/python3.8 -c -c /workspace/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/group_points_gpu.cu -o /workspace/OpenPCDet/build/temp.linux-x86_64-cpython-38/pcdet/ops/pointnet2/pointnet2_stack/src/group_points_gpu.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1013"' -DTORCH_EXTENSION_NAME=pointnet2_stack_cuda -D_GLIBCXX_USE_CXX11_ABI=1 -gencode=arch=compute_52,code=sm_52 -gencode=arch=compute_60,code=sm_60 -gencode=arch=compute_61,code=sm_61 -gencode=arch=compute_70,code=sm_70 -gencode=arch=compute_75,code=sm_75 -gencode=arch=compute_80,code=sm_80 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
[11/13] /usr/local/cuda/bin/nvcc  -I/opt/conda/lib/python3.8/site-packages/torch/include -I/opt/conda/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/include/python3.8 -c -c /workspace/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/vector_pool_gpu.cu -o /workspace/OpenPCDet/build/temp.linux-x86_64-cpython-38/pcdet/ops/pointnet2/pointnet2_stack/src/vector_pool_gpu.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1013"' -DTORCH_EXTENSION_NAME=pointnet2_stack_cuda -D_GLIBCXX_USE_CXX11_ABI=1 -gencode=arch=compute_52,code=sm_52 -gencode=arch=compute_60,code=sm_60 -gencode=arch=compute_61,code=sm_61 -gencode=arch=compute_70,code=sm_70 -gencode=arch=compute_75,code=sm_75 -gencode=arch=compute_80,code=sm_80 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
[12/13] /usr/local/cuda/bin/nvcc  -I/opt/conda/lib/python3.8/site-packages/torch/include -I/opt/conda/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/include/python3.8 -c -c /workspace/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/sampling_gpu.cu -o /workspace/OpenPCDet/build/temp.linux-x86_64-cpython-38/pcdet/ops/pointnet2/pointnet2_stack/src/sampling_gpu.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1013"' -DTORCH_EXTENSION_NAME=pointnet2_stack_cuda -D_GLIBCXX_USE_CXX11_ABI=1 -gencode=arch=compute_52,code=sm_52 -gencode=arch=compute_60,code=sm_60 -gencode=arch=compute_61,code=sm_61 -gencode=arch=compute_70,code=sm_70 -gencode=arch=compute_75,code=sm_75 -gencode=arch=compute_80,code=sm_80 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
[13/13] /usr/local/cuda/bin/nvcc  -I/opt/conda/lib/python3.8/site-packages/torch/include -I/opt/conda/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/include/python3.8 -c -c /workspace/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/voxel_query_gpu.cu -o /workspace/OpenPCDet/build/temp.linux-x86_64-cpython-38/pcdet/ops/pointnet2/pointnet2_stack/src/voxel_query_gpu.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1013"' -DTORCH_EXTENSION_NAME=pointnet2_stack_cuda -D_GLIBCXX_USE_CXX11_ABI=1 -gencode=a [... cut ...]
  - LAPACK is enabled (usually provided by MKL)
  - NNPACK is enabled
  - CPU capability usage: AVX2
  - CUDA Runtime 11.6
  - NVCC architecture flags: -gencode;arch=compute_52,code=sm_52;-gencode;arch=compute_60,code=sm_60;-gencode;arch=compute_61,code=sm_61;-gencode;arch=compute_70,code=sm_70;-gencode;arch=compute_75,code=sm_75;-gencode;arch=compute_80,code=sm_80;-gencode;arch=compute_86,code=sm_86;-gencode;arch=compute_86,code=compute_86
  - CuDNN 8.4
  - Magma 2.5.2

TorchVision: 0.13.0a0
OpenCV: 4.5.5
spconv2.0: True
```

jihanyang, 2022-06-24:
It seems that spconv not already supports CUDA11.6 and I am not sure whether OpenPCDet can support such CUDA version. Maybe I will suggest to downgrade the CUDA to 11.1

ammaryasirnaich, 2022-06-24:
> It seems that spconv not already supports CUDA11.6 and I am not sure whether OpenPCDet can support such CUDA version. Maybe I will suggest to downgrade the CUDA to 11.1

i see, I will try to use it with CUDA 11.1 and see how things unfold. Thanks for the response @jihanyang

SanghyunPark01, 2022-06-24:
pytorch version under 1.10.

ammaryasirnaich, 2022-06-24:
> pytorch version under 1.10.

Many thanks @SanghyunPark01 it  work

closed 2022-07-04 by ammaryasirnaich

leducnguyen21, 2022-07-09:
What version of CUDA and Pytorch are you using at the end? I also has problem like you. It could comes from the compatible of pytorch and CUDA right?

ammaryasirnaich, 2022-07-10:
please visit [1016](https://github.com/open-mmlab/OpenPCDet/issues/1016) for this issue

maxiuw, 2023-07-14:
same issue. I also had a problem after updating Cuda to 11.8. Solved by:
Downgraded it to 11.6, new env conda -> python 3.8, torch version '1.10.0+cu111', the project [ST3D](https://github.com/CVMI-Lab/ST3D) also required spconv (I installed spconv-cu116). Everything was built with some warning but without errors.

qqcx, 2023-08-16:
THCudaCheck   ->  AT_CUDA_CHECK

maxiuw, 2023-08-16:
@qqcx just changing the name? In which files?

lacie-life, 2023-10-05:
I met the same issue and this was solved with [this](https://stackoverflow.com/questions/72988735/replacing-thc-thc-h-module-to-aten-aten-h-module)

gitouni, 2024-03-18:
> I met the same issue and this was solved with [this](https://stackoverflow.com/questions/72988735/replacing-thc-thc-h-module-to-aten-aten-h-module)

Thanks @lacie-life . This solution works for torch 1.13.1 + cuda 11.6