open-mmlab/OpenPCDet #1015: Attempt to run OpenPCDet demo on GOOGLE Colab platform, but RunERROR

Cappuccinoll, 2022-06-26:
Hi, 

i want to try demo.py on Google Colab platform. All is perfect but when i want to install 'pcdet' library
`python setup.py develop`
i meet an error, this 
-------------------------------------------
/usr/local/lib/python3.7/dist-packages/setuptools/dist.py:484: UserWarning: Normalizing '0.5.2+0000000' to '0.5.2+0'
  warnings.warn(tmpl.format(**locals()))
running develop
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
reading manifest file 'pcdet.egg-info/SOURCES.txt'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
building 'content.OpenPCDet.pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
Emitting ninja build file /content/build/temp.linux-x86_64-3.7/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: error: '/content/content/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp', needed by '/content/build/temp.linux-x86_64-3.7/content/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_cpu.o', missing and no known rule to make it
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 1673, in _run_ninja_build
    env=env)
  File "/usr/lib/python3.7/subprocess.py", line 512, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['ninja', '-v']' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/content/OpenPCDet/setup.py", line 116, in <module>
    'src/sampling_gpu.cu',
  File "/usr/local/lib/python3.7/dist-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.7/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/develop.py", line 34, in run
    self.install_for_development()
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/develop.py", line 136, in install_for_development
    self.run_command('build_ext')
  File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/build_ext.py", line 79, in run
    _build_ext.run(self)
  File "/usr/local/lib/python3.7/dist-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/usr/lib/python3.7/distutils/command/build_ext.py", line 340, in run
    self.build_extensions()
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 708, in build_extensions
    build_ext.build_extensions(self)
  File "/usr/local/lib/python3.7/dist-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/usr/lib/python3.7/distutils/command/build_ext.py", line 449, in build_extensions
    self._build_extensions_serial()
  File "/usr/lib/python3.7/distutils/command/build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/build_ext.py", line 202, in build_extension
    _build_ext.build_extension(self, ext)
  File "/usr/lib/python3.7/distutils/command/build_ext.py", line 534, in build_extension
    depends=ext.depends)
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 538, in unix_wrap_ninja_compile
    with_cuda=with_cuda)
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 1359, in _write_ninja_file_and_compile_objects
    error_prefix='Error compiling objects for extension')
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 1683, in _run_ninja_build
    raise RuntimeError(message) from e
RuntimeError: Error compiling objects for extension
-------------------------------------------

my version (default version of colab):
pytorch 1.11.0
cuda 11.3
cmake 3.22.5
python 3.7.13

does someone have the same problem and how to solve that? i try a lot of version combinations but have no effect.

Thanks!

leducnguyen21, 2022-07-09:
Have you successfully installed ? I also want to run in Google Colab but it shows that google colab don't support the visualization tools so it can't show the image with 3d object detection bouding box. With the setup.py I can run on Google Colab but now I want to run it locally on VSCode but also got the same error as you are. I think it is about Pytorch and CUDA compatible version problem. What version are you using?

OrangeSodahub, 2022-07-20:
Hello, I've occurred this before and I solved it. You can refer to README of https://github.com/OrangeSodahub/CRLFnet#install--build In the part of `INSTALL & BUILD`

closed 2022-09-03 by github-actions[bot]