open-mmlab/OpenPCDet #1017: The size of tensor a (12160) must match the size of tensor b (199680) at non-singleton dimension 1

KyleYueye, 2022-07-03:
I've met a problem, have you ever delt with it?

> Traceback (most recent call last):                                                          | 0/3400 [00:00<?, ?it/s]
  File "network.py", line 116, in <module>
    train_model(
  File "/home/yueye/code/3D-MAN/tools/train_utils/train_utils.py", line 84, in train_model
    accumulated_iter = train_one_epoch(
  File "/home/yueye/code/3D-MAN/tools/train_utils/train_utils.py", line 36, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/home/yueye/code/3D-MAN/pcdet/models/__init__.py", line 42, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/home/yueye/anaconda3/envs/3dman/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/yueye/code/3D-MAN/pcdet/models/detectors/pointpillar.py", line 14, in forward
    loss, tb_dict, disp_dict = self.get_training_loss()
  File "/home/yueye/code/3D-MAN/pcdet/models/detectors/pointpillar.py", line 27, in get_training_loss
    loss_rpn, tb_dict = self.dense_head.get_loss()
  File "/home/yueye/code/3D-MAN/pcdet/models/dense_heads/anchor_head_template.py", line 217, in get_loss
    cls_loss, tb_dict = self.get_cls_layer_loss()
  File "/home/yueye/code/3D-MAN/pcdet/models/dense_heads/anchor_head_template.py", line 128, in get_cls_layer_loss
    cls_loss_src = self.cls_loss_func(cls_preds, one_hot_targets, weights=cls_weights)  # [N, M]
  File "/home/yueye/anaconda3/envs/3dman/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  **File "/home/yueye/code/3D-MAN/pcdet/utils/loss_utils.py", line 59, in forward**
    pt = target * (1.0 - pred_sigmoid) + (1.0 - target) * pred_sigmoid
**RuntimeError: The size of tensor a (12160) must match the size of tensor b (199680) at non-singleton dimension 1**

jihanyang, 2022-07-04:
Have you modified the detection range or voxel size?

KyleYueye, 2022-07-04:
> Have you modified the detection range or voxel size?

I use this model on my own dataset, i've modified the pointcloud range in model yaml config file, voxel size is also modified. But I'm not sure the voxel size is appropriate. How can I calculate the suitable voxel size for my dataset? Thank you.

KyleYueye, 2022-07-04:
> Have you modified the detection range or voxel size?

And now I can locate the bug is happened at file "pcdet/models/dense_heads/anchor_head_template.py", in function `get_cls_layer_loss()`. The shape of `cls_preds` is `torch.Size([4, 645120, 4])` while other tensors are all `torch.Size([4, 40320])` or `torch.Size([4, 40320, 4])`

But I still don't know the reason why this could happen.

jihanyang, 2022-07-04:
Please follow issue about custom dataset to modify your point range and voxel size.

closed 2022-07-04 by jihanyang

s1mpleee, 2022-11-08:
try change feature_map_stride in ANCHOR_GENERATOR_CONFIG