open-mmlab/OpenPCDet #1023: How to visualize point cloud when running demo.py with pretrain model

leducnguyen21, 2022-07-06:
This is my code

%cd /content/OpenPCDet/tools
! python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml \
    --ckpt /content/OpenPCDet/data/kitti/pv_rcnn_8369.pth \
    --data_path /content/OpenPCDet/data/kitti/training/velodyne/000000.bin

This is the error, i don't know how to fix
/content/OpenPCDet/tools
2022-07-06 13:12:58,075   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2022-07-06 13:12:58,075   INFO  Total number of samples: 	1
2022-07-06 13:13:01,908   INFO  ==> Loading parameters from checkpoint /content/OpenPCDet/data/kitti/pv_rcnn_8369.pth to CPU
2022-07-06 13:13:01,978   INFO  ==> Done (loaded 367/367)
2022-07-06 13:13:02,311   INFO  Visualized sample index: 	1
[Exception|implicit_gemm_pair]indices=torch.Size([0, 4]),bs=1,ss=[41, 1600, 1408],algo=ConvAlgo.MaskImplicitGemm,ksize=[3, 3, 3],stride=[1, 1, 1],padding=[1, 1, 1],dilation=[1, 1, 1],subm=True,transpose=False
SPCONV_DEBUG_SAVE_PATH not found, you can specify SPCONV_DEBUG_SAVE_PATH as debug data save path to save debug data which can be attached in a issue.
Traceback (most recent call last):
  File "demo.py", line 125, in <module>
    main()
  File "demo.py", line 109, in main
    pred_dicts, _ = model.forward(data_dict)
  File "../pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/usr/local/lib/python3.7/dist-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "../pcdet/models/backbones_3d/spconv_backbone.py", line 148, in forward
    x = self.conv_input(input_sp_tensor)
  File "/usr/local/lib/python3.7/dist-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/spconv/pytorch/modules.py", line 137, in forward
    input = module(input)
  File "/usr/local/lib/python3.7/dist-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/spconv/pytorch/conv.py", line 393, in forward
    raise e 
  File "/usr/local/lib/python3.7/dist-packages/spconv/pytorch/conv.py", line 384, in forward
    timer=input._timer)
  File "/usr/local/lib/python3.7/dist-packages/spconv/pytorch/ops.py", line 340, in get_indice_pairs_implicit_gemm
    stream_int=stream)
RuntimeError: /tmp/pip-build-env-s_j4zkty/overlay/lib/python3.7/site-packages/cumm/include/tensorview/cuda/launch.h(53)
N > 0 assert faild. CUDA kernel launch blocks must be positive, but got N= 0

aldipiroli, 2022-07-20:
This might be due to a wrong spconv installation. 
Can you try and run the demo script inside the official docker container?

OrangeSodahub, 2022-07-20:
Maybe this is because that your dataset contains empty file (no valid object)

closed 2022-09-03 by github-actions[bot]