open-mmlab/OpenPCDet #1026: Running python setup.py develop

Joaovsky, 2022-07-12:
No CUDA runtime is found, using CUDA_HOME='/usr/local/cuda-11.3'
running develop
/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/setuptools/command/easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
reading manifest file 'pcdet.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
Traceback (most recent call last):
  File "setup.py", line 34, in <module>
    setup(
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/setuptools/__init__.py", line 87, in setup
    return distutils.core.setup(**attrs)
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 148, in setup
    return run_commands(dist)
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 163, in run_commands
    dist.run_commands()
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 967, in run_commands
    self.run_command(cmd)
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/setuptools/dist.py", line 1214, in run_command
    super().run_command(command)
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
    cmd_obj.run()
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/setuptools/command/develop.py", line 34, in run
    self.install_for_development()
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/setuptools/command/develop.py", line 114, in install_for_development
    self.run_command('build_ext')
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/setuptools/dist.py", line 1214, in run_command
    super().run_command(command)
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
    cmd_obj.run()
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 79, in run
    _build_ext.run(self)
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 406, in build_extensions
    self._check_cuda_version()
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 782, in _check_cuda_version
    torch_cuda_version = packaging.version.parse(torch.version.cuda)
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/pkg_resources/_vendor/packaging/version.py", line 49, in parse
    return Version(version)
  File "/home/mola/anaconda3/envs/testtorch/lib/python3.8/site-packages/pkg_resources/_vendor/packaging/version.py", line 264, in __init__
    match = self._regex.search(version)
TypeError: expected string or bytes-like object

Joaovsky, 2022-07-13:
Solved it by having only one spconv in my python environment (version 11.3), instead of having versions for cuda 102 e 113.
Also, while I was debugging, I added absolute path for pcdet/version.py, in target file (line 32, setup.py).

yimingshan666, 2022-08-01:
Hi, I have the same issue. How did you solve this problem?

closed 2022-09-15 by github-actions[bot]

Do-woon, 2023-02-01:
I have the same issue. Please share the solution if someone has.

suryajayaraman, 2023-02-18:
Hi @Do-woon, @Joaovsky, Do you have CUDA and CUDNN installed in the system? I'm trying to install the package in CPU? Do you have any idea if its feasible?

suryajayaraman, 2023-03-21:
Hi @Do-woon , thanks for the reply. By any chance, have you tried installing the package with CPU alone? 

Anyways, I'll try using the pip install command you shared.

Thanks once again for the suggestion

Do-woon, 2023-03-22:
@suryajayaraman No, I only tried GPU version