open-mmlab/OpenPCDet #1027: FileNotFoundError: The file for pre-train model cannot fild

leducnguyen21, 2022-07-13:
I want to know after download pretrain model where the file should locate. I keep getting the message can't find the file following run the code in demo.py. It may stuck at --ckpt pv_rcnn_8369.pth even I already put pv_rcnn_8369.pth in the tools directory
python demo.py 
--cfg_file cfgs/kitti_models/pv_rcnn.yaml 
--ckpt pv_rcnn_8369.pth 
--data_path ../data/kitti/testing/velodyne/000001.bin

The code is as above

leducnguyen21, 2022-07-13:
2022-07-13 10:03:16,171   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2022-07-13 10:03:16,172   INFO  Total number of samples:        1
Traceback (most recent call last):
  File "demo.py", line 112, in <module>
    main()
  File "demo.py", line 90, in main
    model.load_params_from_file(filename=args.ckpt, logger=logger, to_cpu=True)
  File "/home/nguyenle/OpenPCDet/pcdet/models/detectors/detector3d_template.py", line 363, in load_params_from_file
    raise FileNotFoundError
FileNotFoundError

jihanyang, 2022-07-14:
Your path is not correct.

closed 2022-07-14 by jihanyang