open-mmlab/OpenPCDet #1029: why pointpillars does not use spconv?

Sukbin-Lim, 2022-07-14:
is there any reason that why pointpillars does not use sponv?
Does it drop accuracy?

daeunni, 2022-07-15:
Hi, 
please refer this original [pointpillar paper](https://openaccess.thecvf.com/content_CVPR_2019/papers/Lang_PointPillars_Fast_Encoders_for_Object_Detection_From_Point_Clouds_CVPR_2019_paper.pdf)

OrangeSodahub, 2022-07-18:
Any more information?

Sukbin-Lim, 2022-07-18:
When I trained PointPillars with default setting on OpenPCDet.
I get this result with 3 classes of training.
Car AP@0.70, 0.70, 0.70:
bev  AP:89.2605, 86.2989, 83.0038
3d   AP:85.6227, 75.6641, 71.7263

However, there is a difference of about +5 mAP compared to the paper's result.

So, I wonder that the different of mAP came from whether or not to use spconv (The Github of PointPiilars uses SECOND's backbone)

closed 2022-09-02 by github-actions[bot]

Townjj, 2022-09-13:
so why the improvment?