open-mmlab/OpenPCDet #1037: coordinate transformation from left-hand rule to openpcdet local system

ChunmianLin, 2022-07-22:
Hi, how should I transform my own lidar data based on left-hand rule coordinate to the openpcdet local system?

ChunmianLin, 2022-07-22:
Thanks, but it does not follow the kitti format. So how should I transform it?

ChunmianLin, 2022-07-22:
By the way, does the kitti coordinate follow the left-hand rule?

OrangeSodahub, 2022-07-22:
> By the way, does the kitti coordinate follow the left-hand rule?

I mean, only several formats OpenPCDet support like kitti, nuscenes... You could transform your coordinates to kitti coordiantes and proceed. Above link shows an example of how to transform and it may need another function.

ChunmianLin, 2022-07-22:
Many thanks. I am not sure whether my data could be transformed into the kitti format, and am confused about the coordinate transformation.---- Replied Message ***@***.***>Date07/22/2022 21:59 ***@***.***> ***@***.******@***.***>SubjectRe: [open-mmlab/OpenPCDet] coordinate transformation from left-hand rule to openpcdet local system (Issue #1037)

By the way, does the kitti coordinate follow the left-hand rule?

I mean, only several formats OpenPCDet support like kitti, nuscenes... You could transform your coordinates to kitti coordiantes and proceed. Above link shows an example of how to transform and it may need another function.

[
{
"target": "https://github.com/open-mmlab/OpenPCDet/issues/1037#issuecomment-1192605402",
"url": "https://github.com/open-mmlab/OpenPCDet/issues/1037#issuecomment-1192605402",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

OrangeSodahub, 2022-07-24:
?
Only you know your own data.

ChunmianLin, 2022-07-24:
Yep. My LiDAR data is generated from Carla simulator.---- Replied Message ***@***.***>Date07/24/2022 10:37 ***@***.***> ***@***.******@***.***>SubjectRe: [open-mmlab/OpenPCDet] coordinate transformation from left-hand rule to openpcdet local system (Issue #1037)
?
Only you know your own data.

[
{
"target": "https://github.com/open-mmlab/OpenPCDet/issues/1037#issuecomment-1193233324",
"url": "https://github.com/open-mmlab/OpenPCDet/issues/1037#issuecomment-1193233324",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

closed 2022-09-08 by github-actions[bot]