open-mmlab/OpenPCDet #1039: How to see both the prediction boxes and GT boxes in demo.py?

HuangZhe885, 2022-07-22:
I noticed the def draw_scenes() in visualize_utils.py ,but how could I write in demo.py ?  I really don`t know how to insert,
Could anybody have the complete code ? thank you soooooo much ,  orz!

OrangeSodahub, 2022-07-22:
In `Demo.py` here input the `pred_dict` (format: [[x, y, x, l, w, h, score],...]) to the draw function to visualize the predict results:
https://github.com/open-mmlab/OpenPCDet/blob/c233477a4a4bf2861a978794d6b9c047f64641f7/tools/demo.py#L98-L103
And you could write another function here to read gt boxes from label files (in the same form of `pred_dict`) and input to the draw function.
It is not difficult.

HuangZhe885, 2022-07-23:
> data_dict in demo.py does not have the key of "gt_boxes".
Below is the code I have added
-----------------------------
# get GT box

print(data_dict['gt_boxes'].shape)
V.draw_scenes(
                points=data_dict['points'][:, 1:], ref_boxes=data_dict['gt_boxes'][0],
                ref_scores=None, ref_labels=None
            )

HuangZhe885, 2022-07-23:
> In `Demo.py` here input the `pred_dict` (format: [[x, y, x, l, w, h, score],...]) to the draw function to visualize the predict results: [...]

data_dict in demo.py does not have the key of "gt_boxes".
Below is the code I have added
---------------------
print(data_dict['gt_boxes'].shape)
V.draw_scenes(
points=data_dict['points'][:, 1:], ref_boxes=data_dict['gt_boxes'][0],
ref_scores=None, ref_labels=None
)

HuangZhe885, 2022-07-23:
> In `Demo.py` here input the `pred_dict` (format: [[x, y, x, l, w, h, score],...]) to the draw function to visualize the predict results: [...]

>  data_dict in `demo.py` does not have the key of "`gt_boxes`".
>  Below is the code I have added

print(data_dict['gt_boxes'].shape)
V.draw_scenes(
points=data_dict['points'][:, 1:], ref_boxes=data_dict['gt_boxes'][0],
ref_scores=None, ref_labels=None
)

OrangeSodahub, 2022-07-23:
@HuangZhe885  You should understand these code more. What I suggested before is not using `gt_boxes` directly from `data_dict`, `data_dict` of course doesn't contain `gt_boxes`, `pred_dict` neither.
Notice that four parameters input to function `V.draw_scene`, what you should do is replacing `pred_boxes` with `gt_boxes` and `gt_boxes` need your manual import ! You can refer to similar codes:
https://github.com/open-mmlab/OpenPCDet/blob/c233477a4a4bf2861a978794d6b9c047f64641f7/pcdet/datasets/kitti/kitti_dataset.py#L245
Relative codes showed the generation of `gt_boxes`. Lastly, make sure that your gt_boxes has the same format as the `pred_dicts` and input them into `V.draw`.

closed 2022-09-07 by github-actions[bot]

Petros626, 2024-10-04:
@OrangeSodahub Why the ground truth labels aren't in the data_dict? I think they are:
[image]

Or maybe we speak about different things?

OrangeSodahub, 2024-10-04:
@Petros626 Two years passed since I saw this issue, I forgot everything ...

How did you make your figure? By hand?

Petros626, 2024-10-04:
hahah I know you're an OpenPCDet veteran here and I've to explore everything new, but I'll leave some good information after I finished my project 🗡️ .

No this from a Power Point presentation from the University itself. It helps to understand OpenPCDet better in some cases.
[OpenPCDet_Tutorial.pdf](https://github.com/user-attachments/files/17261688/OpenPCDet_Tutorial.pdf)

Petros626, 2024-11-07:
> In `Demo.py` here input the `pred_dict` (format: [[x, y, x, l, w, h, score],...]) to the draw function to visualize the predict results: [...]

for drawing a scene are the labels in camera coordinates or already transformed lidar coordinates necessary? I get weird looking bounding boxes.