open-mmlab/OpenPCDet #1049: Bug in database sampler

pk1996, 2022-08-02:
Hello! 

Thank you for this wonderful codebase. 

I believe there is an error in the database sampler here - https://github.com/open-mmlab/OpenPCDet/blob/c233477a4a4bf2861a978794d6b9c047f64641f7/pcdet/datasets/augmentor/database_sampler.py#L387-L388

This causes issues in cases where the user is loading data encoded in other data formats (like float64) leading to NaN errors as captured here - https://github.com/open-mmlab/OpenPCDet/issues/341. A check needs to be kept in place to ensure the dataset class loads the point cloud in the specified format (float32 I presume).

Thanks!

jihanyang, 2022-08-03:
Thanks for your feedback. Would you like to pose an example and launch a PR to fix this problem?

pk1996, 2022-08-04:
Sure I can give it a try!

closed 2022-09-19 by github-actions[bot]