open-mmlab/OpenPCDet #1051: Evaluation and Train code

pk1996, 2022-08-08:
Hi 

I am raising this as an issue but it's more of a bug/question to the authors of this great code base!

So I observe that the train code is ordered such that it first does the assigned training epochs and then goes for the evaluation code for the last K (passed via args) checkpoints. 

I would have preferred evaluation to be interleaved periodically with training in order to get a good idea of the eval trend. Secondly, I believe this will help in identifying the best model better. In the current setup, we are in a way assuming that the best model exists in the last K epochs which might not be true. 

TLDR - Have eval code run in periodically after every K epochs of training. Save the checkpoint here and also evaluate the best model in these eval runs. All this instead of having it at the end.

I might have missed this feature so in that case kindly point me to it. Else I can probably try adding this and raise a PR! 

Thanks

jihanyang, 2022-08-09:
Hello, we only support `--eval_all` to evaluate all saved check points in `test.py`. We haven't supported any features to evaluate along with training procedure in `train.py`.

pk1996, 2022-08-09:
Thank you for your reply. 
Is there any specific reason for this?

jihanyang, 2022-08-09:
On the one hand, some 3D detection datasets are large, switching between training and validation set might take a lot of time. On the other hand, we typically only validate the last epoch or can do evaluation after training, so we don't have urgently demands for such feature.

closed 2022-08-09 by pk1996