open-mmlab/OpenPCDet #1056: probelem of caddn save checkpoints.

zyrant, 2022-08-11:
environment：
```
cuda 10.2
python 3.8
torch 1.9.0+cu102
torchvision 0.10.0+cu102
pcdet 0.5.2
```

```
2022-08-11 09:58:58,734   INFO  **********************Start training cfgs/kitti_models/CaDDN(default)**********************
epochs:   0%|                                                                                                 | 0/1 [00:00<?, ?it/s/home/miniconda3/envs/sslpcdet/lib/python3.8/site-packages/torch/nn/functional.py:718: UserWarning: Named tensors and all their associated APIs are an experimental feature and subject to change. Please do not use them for anything important until they are released as stable. (Triggered internally at  /pytorch/c10/core/TensorImpl.h:1156.)
  return torch.max_pool2d(input, kernel_size, stride, padding, dilation, ceil_mode)
/home/miniconda3/envs/sslpcdet/lib/python3.8/site-packages/torch/nn/functional.py:718: UserWarning: Named tensors and all their associated APIs are an experimental feature and subject to change. Please do not use them for anything important until they are released as stable. (Triggered internally at  /pytorch/c10/core/TensorImpl.h:1156.)
  return torch.max_pool2d(input, kernel_size, stride, padding, dilation, ceil_mode)
train: 100%|████████████████████████████████████████████████████████████████████████| 928/928 [20:52<00:00,  1.35s/it, total_it=928]
Traceback (most recent call last):
  File "tools/train.py", line 203, in <module>
    main()
  File "tools/train.py", line 193, in main
    repeat_eval_ckpt(
  File "/opt/data/private/PCDet/tools/test.py", line 115, in repeat_eval_ckpt
    model.load_params_from_file(filename=cur_ckpt, logger=logger, to_cpu=dist_test)
  File "/opt/data/private/PCDet/pcdet/models/detectors/detector3d_template.py", line 367, in load_params_from_file
    checkpoint = torch.load(filename, map_location=loc_type)
  File "/home/miniconda3/envs/sslpcdet/lib/python3.8/site-packages/torch/serialization.py", line 608, in load
    return _legacy_load(opened_file, map_location, pickle_module, **pickle_load_args)
  File "/home/miniconda3/envs/sslpcdet/lib/python3.8/site-packages/torch/serialization.py", line 794, in _legacy_load
    deserialized_objects[key]._set_from_file(f, offset, f_should_read_directly)
RuntimeError: unexpected EOF, expected 1024 more bytes. The file might be corrupted.

```
when i set the **NUM_EPOCHS : 1** in CaDDN.yaml, it can not save the checkpoint correctly. 
the saved checkpoint is only about 115M, but actually, the checkpoint should be about 700M.
I tried to set the epoch to 2, the problem happened in the last round, when the epoch was 2, the weight could not be saved properly.
PS: this problem does not occur with Pointpillar. yaml.

closed 2022-08-15 by zyrant

jlqzzz, 2022-09-20:
> environment：
> 
> ```
> cuda 10.2
> python 3.8
> torch 1.9.0+cu102
> torchvision 0.10.0+cu102
> pcdet 0.5.2
> ```
> 
> ```
> 2022-08-11 09:58:58,734   INFO  **********************Start training cfgs/kitti_models/CaDDN(default)**********************
> epochs:   0%|                                                                                                 | 0/1 [00:00<?, ?it/s/home/miniconda3/envs/sslpcdet/lib/python3.8/site-packages/torch/nn/functional.py:718: UserWarning: Named tensors and all their associated APIs are an experimental feature and subject to change. Please do not use them for anything important until they are released as stable. (Triggered internally at  /pytorch/c10/core/TensorImpl.h:1156.)
>   return torch.max_pool2d(input, kernel_size, stride, padding, dilation, ceil_mode)
> /home/miniconda3/envs/sslpcdet/lib/python3.8/site-packages/torch/nn/functional.py:718: UserWarning: Named tensors and all their associated APIs are an experimental feature and subject to change. Please do not use them for anything important until they are released as stable. (Triggered internally at  /pytorch/c10/core/TensorImpl.h:1156.)
>   return torch.max_pool2d(input, kernel_size, stride, padding, dilation, ceil_mode)
> train: 100%|████████████████████████████████████████████████████████████████████████| 928/928 [20:52<00:00,  1.35s/it, total_it=928]
> Traceback (most recent call last):
>   File "tools/train.py", line 203, in <module>
>     main()
>   File "tools/train.py", line 193, in main
>     repeat_eval_ckpt(
>   File "/opt/data/private/PCDet/tools/test.py", line 115, in repeat_eval_ckpt
>     model.load_params_from_file(filename=cur_ckpt, logger=logger, to_cpu=dist_test)
>   File "/opt/data/private/PCDet/pcdet/models/detectors/detector3d_template.py", line 367, in load_params_from_file
>     checkpoint = torch.load(filename, map_location=loc_type)
>   File "/home/miniconda3/envs/sslpcdet/lib/python3.8/site-packages/torch/serialization.py", line 608, in load
>     return _legacy_load(opened_file, map_location, pickle_module, **pickle_load_args)
>   File "/home/miniconda3/envs/sslpcdet/lib/python3.8/site-packages/torch/serialization.py", line 794, in _legacy_load
>     deserialized_objects[key]._set_from_file(f, offset, f_should_read_directly)
> RuntimeError: unexpected EOF, expected 1024 more bytes. The file might be corrupted.
> ```
> 
> when i set the **NUM_EPOCHS : 1** in CaDDN.yaml, it can not save the checkpoint correctly. the saved checkpoint is only about 115M, but actually, the checkpoint should be about 700M. I tried to set the epoch to 2, the problem happened in the last round. PS: this problem does not occur with Pointpillar. yaml.
> 
> THX.

Hello, How did you solve this problem?

YWonchall, 2023-11-09:
> Hello, How did you solve this problem?

hi, how did you solve this problem?