open-mmlab/OpenPCDet #1059: Different results from original paper on kitti test set (PV-RCNN++)

medlen, 2022-08-16:
I saw results listed on table 16(PV-RCNN++) is different from original paper. Are the results reproduced here?

sshaoshuai, 2022-08-20:
The results in this repo are reported on validation set, and the codes of original paper are almost same with this repo, yet training with different times may have variations due to the relatively smaller dataset.

JLIN77, 2022-09-20:
Yes. I did inference the KITTI test set with official PV-RCNN weight, and submit the results to KITTI website. The results show as:
      Car (3D Detection) | 66.01 % | 64.26 % | 61.65 %
      Pedestrian (3D Detection) | 35.58 % | 29.18 % | 27.36 %
      Cyclist (3D Detection) | 51.01 % | 42.79 % | 39.31 %
Compared with mmlab-PVRCNN model on KITTI leaderboard, it drops almost 10~20 mAP. 
Any answer to this question?

closed 2022-10-05 by github-actions[bot]