open-mmlab/OpenPCDet #1060: Why does using ground truth as prediction does not yield 100%?

marcelbrucker, 2022-08-16:
Hi,
testing your evaluation with a random subset of the ground truth from KITTI as prediction yields strange results, far from the expected 100% AP.
On some random two frames I obtain this:
```
('Car AP@0.70, 0.70, 0.70:\n'
 'bbox AP:9.0909, 9.0909, 9.0909\n'
 'bev  AP:0.0000, 0.0000, 0.0000\n'
 '3d   AP:0.0000, 0.0000, 0.0000\n'
 'aos  AP:9.09, 9.09, 9.09\n'
 'Car AP_R40@0.70, 0.70, 0.70:\n'
 'bbox AP:2.5000, 2.5000, 5.0000\n'
 'bev  AP:0.0000, 0.0000, 0.0000\n'
 '3d   AP:0.0000, 0.0000, 0.0000\n'
 'aos  AP:2.50, 2.50, 5.00\n'
 'Car AP@0.70, 0.50, 0.50:\n'
 'bbox AP:9.0909, 9.0909, 9.0909\n'
 'bev  AP:0.0000, 0.0000, 0.0000\n'
 '3d   AP:0.0000, 0.0000, 0.0000\n'
 'aos  AP:9.09, 9.09, 9.09\n'
 'Car AP_R40@0.70, 0.50, 0.50:\n'
 'bbox AP:2.5000, 2.5000, 5.0000\n'
 'bev  AP:0.0000, 0.0000, 0.0000\n'
 '3d   AP:0.0000, 0.0000, 0.0000\n'
 'aos  AP:2.50, 2.50, 5.00\n',
 {'Car_3d/easy_R40': 0.0,
  'Car_3d/hard_R40': 0.0,
  'Car_3d/moderate_R40': 0.0,
  'Car_aos/easy_R40': 2.5,
  'Car_aos/hard_R40': 5.0,
  'Car_aos/moderate_R40': 2.5,
  'Car_bev/easy_R40': 0.0,
  'Car_bev/hard_R40': 0.0,
  'Car_bev/moderate_R40': 0.0,
  'Car_image/easy_R40': 2.5,
  'Car_image/hard_R40': 5.0,
  'Car_image/moderate_R40': 2.5})
```

I use the suggested python script from the [README.md](https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/kitti/kitti_object_eval_python/README.md) and two original KITTI labels as ground truth as well as as prediction.
Some quick reproduction of my computation can be done using the [file](https://github.com/marcelbrucker/OpenPCDet/blob/master/pcdet/datasets/kitti/kitti_object_eval_python/evaluate_a9-dataset.py) from this [commit](https://github.com/marcelbrucker/OpenPCDet/commit/d5a3ae741fc79712d55f62e09086ee28755878e5) from my fork, where I included the used subset of the ground truth.

I'd be happy to understand why I don't simply obtain 100% for all detection.

marcelbrucker, 2022-08-17:
> 3D detector cannot overfit to a small subset without some specific modification. Just try to validate on the whole training set.

I am not using any detector. I just want to see that the evaluation works as expected by using the ground truth as my predictions.

yimingshan666, 2022-08-29:
I met the same issue, did you find out what caused it?

marcelbrucker, 2022-08-30:
> I met the same issue, did you find out what caused it?

Unfortunately not, I started using another evaluation kit.

yimingshan666, 2022-08-30:
> 

Have you increased the amount of data for evaluation, like using 100 frames instead of 1 frame. I am trying.

closed 2022-10-15 by github-actions[bot]