open-mmlab/OpenPCDet #1062: ONCE benchmark support

AvivSham, 2022-08-17:
Hi All,
Thank you for this wonderful repo!
We want to use OpenPCDet with ONCE dataset, do you plan to support it? If not do you have documentation / tutorial how to incorporate custom dataset?
Basically we want to do some domain adaptation research with point clouds so if the current supported datasets has domain adaptation splits it should be fine. Do you know if any of them (lyft / Kitti, etc.) have such splits?

Cheers,
A

jihanyang, 2022-08-18:
Have you 3D detection DA works such as ST3D? Also, as we know, ONCE benchmark already provide DA configs based on OpenPCDet, so what's the main issue?

AvivSham, 2022-08-18:
I did not see that they released config files based on OpenPCDet (can you please share the link?). If that is the case than you can close this issue and if anything else comes up I will reopen. 
Basically my issue is with using ONCE with OpenPCDet I'm facing problems with CUDA recompiling, but I just saw there is Docker support so the best way would be to try and use it.

closed 2022-10-02 by github-actions[bot]