open-mmlab/OpenPCDet #1066: Adding new classes to a pre-trained model

d33dler, 2022-08-18:
Assuming the model was trained on kitti/waymo dataset with 3 classes, can we add custom new classes and continue training with addition of these new classes? (by changing the dataset & model configs, specifically PVRCNN++) 
Thanks
@sshaoshuai

jihanyang, 2022-08-19:
You can have a try. But as the cls head will change with more class, I am not sure how the performance will behave.

d33dler, 2022-08-19:
> You can have a try. But as the cls head will change with more class, I am not sure how the performance will behave.

@jihanyang So it is preferred to train with all classes from the beginning? Approximately how many classes could be used until the performance begins to degrade?

d33dler, 2022-08-22:
@jihanyang some insight would be greatly appreciated!

closed 2022-10-07 by github-actions[bot]