open-mmlab/OpenPCDet #1070: Any 3DSSD config yaml suggestion in Waymo dataset?

regretBigShark, 2022-08-22:
Hi, first I really appreciate your team's excellent work on 3D point-cloud object detection. And also, thanks to https://github.com/zye1996/3DSSD-torch and https://github.com/qiqihaer/3DSSD-pytorch-openPCDet, I have edit 3DSSD config yaml on Waymo dataset like following:
```
CLASS_NAMES: ['Vehicle', 'Pedestrian', 'Cyclist']

DATA_CONFIG:
  _BASE_CONFIG_: cfgs/dataset_configs/waymo_dataset.yaml

  DATA_PROCESSOR:
    -   NAME: mask_points_and_boxes_outside_range
        REMOVE_OUTSIDE_BOXES: True

    -   NAME: sample_points
        NUM_POINTS: {
          'train': 16384,
          'test': 16384
        }

    -   NAME: shuffle_points
        SHUFFLE_ENABLED: {
          'train': True,
          'test': False
        }

MODEL:
  NAME: SSD3D

  BACKBONE_3D:
    NAME: PointNet2MSG_FPS
    SA_CONFIG:
      NPOINTS: [[4096], [512], [256, 256]]
      RADIUS: [[0.2,0.4,0.8], [0.4,0.8,1.6], [1.6,3.2,4.8]]
      NSAMPLE: [[32,32,64], [32,32,64], [32,32,32]]
      MLPS: [[[ 16,16,32 ], [ 16,16,32 ], [ 32,32,64]],
             [[ 64,64,128 ], [ 64,64,128 ], [ 64,96,128]],
             [[ 128,128,256 ], [ 128,192,256 ], [ 128,256,256 ]]]
      DILATED_GROUP: [True, True, True]
      FPS_MOD: [['D-FPS'], ['FS'], ['F-FPS', 'D-FPS']]
      FPS_SAMPLE_RANGE: [[-1], [-1], [512, -1]]
    AGGREGATION_CHANNELS: [64, 128, 256]
    FP_MLPS: []

  PFE:
    NAME: VoteModule
    NUM_POINTS: [256]
    NUM_INPUT_FEATURES: 256
    MLPS: [128]
    VOTE_RANGE: [3.0, 3.0, 2.0]
    WITH_RES_FEATURE: False
    AGGREGATION_MLPS: [[256, 256, 256, 512], [256, 256, 512, 1024]]
    AGGREGATION_OUT: 512
    AGGREGATION_RADIUS: [4.8, 6.4]
    AGGREGATION_NSAMPLES: [16, 32]

  POINT_HEAD:
    NAME: PointHeadBox3DSSD
    CLS_FC: [ 256, 256 ]
    REG_FC: [ 256, 256 ]
    CLASS_AGNOSTIC: False
    USE_POINT_FEATURES_BEFORE_FUSION: False
    TARGET_CONFIG:
      GT_EXTRA_WIDTH: [ 0.2, 0.2, 0.2 ]
      BOX_CODER: PointResidual_BinOri_Coder
      BOX_CODER_CONFIG: {
        'use_mean_size': True,
        'mean_size': [
          [ 3.9, 1.6, 1.56 ],
          [ 0.8, 0.6, 1.73 ],
          [ 1.76, 0.6, 1.73 ]
        ]
      }

    LOSS_CONFIG:
      LOSS_REG: WeightedSmoothL1Loss
      CORNER_LOSS_REGULARIZATION: True
      CENTERNESS_REGULARIZATION: True
      IOU_REGULARIZATION: False
      POINT_BOX_REG: "l1"
      LOSS_WEIGHTS: {
        'point_cls_weight': 1.0,
        'point_box_weight': 1.0,
        'corner_weight': 1.0,
        'code_weights': [ 1.0, 1.0, 1.0, 1.0, 1.0, 1.0 ]
      }

  POST_PROCESSING:
    RECALL_THRESH_LIST: [ 0.3, 0.5, 0.7 ]
    SCORE_THRESH: 0.1
    OUTPUT_RAW_SCORE: False

    EVAL_METRIC: waymo

    NMS_CONFIG:
      MULTI_CLASSES_NMS: False
      NMS_TYPE: nms_gpu
      NMS_THRESH: 0.1
      NMS_PRE_MAXSIZE: 4096
      NMS_POST_MAXSIZE: 500

OPTIMIZATION:
  BATCH_SIZE_PER_GPU: 12
  NUM_EPOCHS: 100

  OPTIMIZER: adam_onecycle
  LR: 0.003
  WEIGHT_DECAY: 0.01
  MOMENTUM: 0.9

  MOMS: [0.95, 0.85]
  PCT_START: 0.4
  DIV_FACTOR: 10
  DECAY_STEP_LIST: [80, 120]
  LR_DECAY: 0.1
  LR_CLIP: 0.0000001

  LR_WARMUP: False
  WARMUP_EPOCH: 1

  GRAD_NORM_CLIP: 10
```

And here is some information about my training data:

```
2022-08-18 14:02:49,794   INFO  Database filter by min points Vehicle: 1156571 => 986851
2022-08-18 14:02:50,024   INFO  Database filter by min points Pedestrian: 1064234 => 899433
2022-08-18 14:02:50,050   INFO  Database filter by min points Cyclist: 52005 => 46336
2022-08-18 14:02:50,361   INFO  Database filter by difficulty Vehicle: 986851 => 986851
2022-08-18 14:02:50,684   INFO  Database filter by difficulty Pedestrian: 899433 => 899433
2022-08-18 14:02:50,698   INFO  Database filter by difficulty Cyclist: 46336 => 46336
2022-08-18 14:02:51,272   INFO  Loading Waymo dataset
2022-08-18 14:03:41,284   INFO  Total skipped info 25
2022-08-18 14:03:41,284   INFO  Total samples for Waymo dataset: 153125
2022-08-18 14:03:41,293   INFO  Total sampled samples for Waymo dataset: 30625
```

I have trained 100 epochs on 2 RTX 3090 GPUs, and each batch size was 12. Unluckily, the evaluation result was really poor on checkpoint_100.pth. The VEHICLE_LEVEL_1 AP was 0.2945, and the PEDSTRAIN_LEVEL1_1 AP was only 0.0723.

Is there anyone who have trained 3DSSD model on Waymo dataset?

jihanyang, 2022-08-22:
Have you reproduce the performance of 3DSSD on KITTI? I suspect it is cuased by the number of sampled points is not enough for waymo dataset.
```
-   NAME: sample_points
        NUM_POINTS: {
          'train': 16384,
          'test': 16384
        }
```

regretBigShark, 2022-08-22:
> Have you reproduce the performance of 3DSSD on KITTI? I suspect it is cuased by the number of sampled points is not enough for waymo dataset.
> 
> ```
> -   NAME: sample_points
>         NUM_POINTS: {
>           'train': 16384,
>           'test': 16384
>         }
> ```

Hi, thanks for your reply. I have reproduced the performance of 3DSSD on KITTI, and result is almost closed to https://github.com/zye1996/3DSSD-torch.

jihanyang, 2022-08-22:
You can consider to try my suggestion (see another point-based method IA-SSD: https://github.com/yifanzhang713/IA-SSD/blob/main/tools/cfgs/waymo_models/IA-SSD.yaml)

closed 2022-08-22 by jihanyang