open-mmlab/OpenPCDet #1076: Can we Export the model to standard Pytorch Model format ?

ammaryasirnaich, 2022-08-24:
I wanted to know if you can export the models to standard pytorch models ?

sshaoshuai, 2022-09-07:
I am not sure what the problem is. Current the model is saved by native torch.save() with explicit name-param pairs.

ammaryasirnaich, 2022-09-07:
Actually, I mean, can we traverse convolutional layers in the pre-trained model  for Feature Maps as with the below code which can be done with a standard pytorch model 
```
no_of_layers=0
conv_layers=[]
model_children=list(modelVGG.children())
 for child in model_children:
  print(type(child))
  if type(child)==nn.Conv2d:
    no_of_layers+=1
    conv_layers.append(child)
  elif type(child)==nn.Sequential:
    for layer in child.children():
      if type(layer)==nn.Conv2d:
        no_of_layers+=1
        conv_layers.append(layer)
print(no_of_layers)
```

sshaoshuai, 2022-09-07:
I am still not clear about your question. If you want to collect all convolution layers to a list,  I think it is feasible if you are familiar with the model structure and know what type of layers you would like to collect.

closed 2022-10-24 by github-actions[bot]