open-mmlab/OpenPCDet #1084: Voxel RCNN on Nuscenes

wenchened, 2022-08-31:
@CSautier thanks for your question on #693
I have meet the same problem as [#639 ](url), the issue is like this:
`File "/pcdet/models/roi_heads/target_assigner/proposal_target_layer.py", line 148, in sample_rois_for_rcnn batch_gt_of_rois[index] = cur_gt[gt_assignment[sampled_inds]] RuntimeError: The expanded size of the tensor (8) must match the existing size (10) at non-singleton dimension 1. Target sizes: [128, 8]. Tensor sizes: [128, 10].`
You say . `You can do that in the class NuScenesDataset, (line 33), you will also have to do a similar operation for the augmentation in database_sampler.py (line 179).`, 
coudle you give a sample for it?

CSautier, 2022-08-31:
Hi, I advise you to try the solution of suggested in #639 of "adding a key-value pair 'PRED_VELOCITY: False' in the 'DATA_CONFIG' field in the model definition file (.yaml)". This should fix your issue.

wenchened, 2022-09-02:
> 

I have already made this change
thanks a lot

closed 2022-09-02 by wenchened