open-mmlab/OpenPCDet #1095: Porting TransFusion-L from MMDet3D to OpenPCDet

ytan101, 2022-09-06:
Hello, I have implemented TransFusion-L in MMDet3D (https://github.com/XuyangBai/TransFusion). Seeing that OpenPCDet has lesser dependencies, I would like to port TransFusion-L over from its MMDet3D implementation to OpenPCDet.

Will it be feasible to do so?

Thank you

sshaoshuai, 2022-09-07:
Do you mean that you would like to submit a PR to support TransFusion-L also in OpenPCDet? If so, we are happy to accept such a PR if it does not affect the current structure of OpenPCDet. 
One concern is that TransFusion is a multimodal approach, and I am not sure whether it is easy to add it to OpenPCDet since it also has an image branch. 
Nevertheless, we appreciate it if you can submit such a PR.

sshaoshuai, 2022-09-07:
Got it. 
Yes, we would like to keep OpenPCDet self-contained and as simple as possible. For supporting such a TransFusion head, I think you just need to replace the operations in your provided version with the native PyTorch operations (As I do not look at it very carefully, I think generally it should be ok for almost all the operations).

ytan101, 2022-09-08:
Thank you very much, I will try it out.

closed 2022-09-08 by ytan101