open-mmlab/OpenPCDet #1096: mistake in docs/guidelines_of_approaches/mppnet.md

dingluchuan, 2022-09-08:
I think we should use centerpoint_4frames.yaml not mppnet_4frames.yaml when saving the RPN model's prediction results.
bash scripts/dist_test.sh ${NUM_GPUS}  --cfg_file cfgs/waymo_models/mppnet_4frames.yaml \
--ckpt   ../output/waymo_models/centerpoint_4frames/default/ckpt/checkpoint_epoch_36.pth \
--set   DATA_CONFIG.DATA_SPLIT.test train
# val
bash scripts/dist_test.sh ${NUM_GPUS}  --cfg_file cfgs/waymo_models/mppnet_4frames.yaml \
--ckpt   ../output/waymo_models/centerpoint_4frames/default/ckpt/checkpoint_epoch_36.pth \
--set   DATA_CONFIG.DATA_SPLIT.test val

Cedarch, 2022-09-08:
Thanks for your remind,  when saving RPN's result, the config should be centerpoint_4frames.yaml, we will correct this mistake.

sshaoshuai, 2022-09-08:
Solved in https://github.com/open-mmlab/OpenPCDet/pull/1097.

closed 2022-09-08 by sshaoshuai