open-mmlab/OpenPCDet #1099: Different result between paper & PCDet code in kitti dataset (Pointpillar)

Townjj, 2022-09-11:
hello,
i don't konw why the test result of PCDet 's Pointpillar have a little big improvment above the paper.
For example, use the model pth in PCDet have the result  **57.3015, 51.4145, 46.8715** (Pedestrain 3d R40 with threshold 0.5), but the reault in paper is **52.08 43.53 41.49**.
thanks.

jihanyang, 2022-09-13:
This might be caused by origin Pointpillar doesn't use gt_sampling (cannot remember clearly).

Townjj, 2022-09-13:
Thanks for reply.
May be the another reason is the result in Pointpillar paper is submit to the kitti evaluation benchmark and the result in pcdet is evaluate in kitti val set locally? I don't konw the difference in 2 types of 3d/aos result.

closed 2022-10-28 by github-actions[bot]