open-mmlab/OpenPCDet #1100: evaluation performance of PV-RCNN++ on PandaSet dataset

1uk4s, 2022-09-12:
Hi,

i trained PV-RCNN++ on PandaSet dataset (only spinning lidar data) and evaluated for class Car, Pedestrian, Bicycle and Motorcycle. I tried for 30, 50 and 60 epochs.
My best evaluation results are:
Car @ 0.7 IOU:                53 average precision
Car @ 0.5 IOU:                65 average precision
Pedestrian @ 0.5 IOU:     25 average precision
Pedestrian @ 0.3 IOU:    34 average precision
Bicycle @ 0.5 IOU:          0.2 average precision
Motorcycle @ 0.5 IOU:    9 average precision

As you can see the evaluation results for Bicycle and Motorcycle are way more bad than those of Car and Pedestrian. 
Has someone an idea whats the reason for the bad performance on these classes? Maybe it is due to the class imbalance ? I counted the instances of each class in the test set:
Car: ~60000
Pedestrian: ~1600
Bicycle: ~1000
Motorcycle: ~1000

However, in PandaSet paper they provide a baseline with quite good results for class Bicycle: 

I also trained voxel-RCNN on PandaSet dataset and had the same issue.

Any help would be very appreciated

Thanks in advance
Lukas

jihanyang, 2022-09-13:
Have you used gt_sampling?

1uk4s, 2022-09-13:
thanks for reply!

No at the moment not. Does this help to fix the class imbalance?

jihanyang, 2022-09-13:
Yes. I strongly recommand to use it.

1uk4s, 2022-09-16:
@jihanyang I tried data augmentation but it didnt change something.
I used the code in pandaset_dataset.yaml
I think it's not working as it is also commented in line 61:

`DATA_AUGMENTOR:`
`DISABLE_AUG_LIST: ['placeholder']`
`AUG_CONFIG_LIST:`
`# gt sampling not working at the moment`
`- NAME: gt_sampling`

Do you know what the mistake is that I can try to fix it?

1uk4s, 2022-09-19:
Ok i found the issue. When generating the gt database, the sequence number is not included in the name of the gt objects. This causes the stored objects to be overwritten again and again. Inserting the sequence in the name solves the problem

closed 2022-09-19 by 1uk4s

jonathan1919810, 2023-04-14:
Hi,Lukas.What you do to evaluate successfully in pandaset? I try the api in kitti and fail. Can you give us some hint or some code.

JuanIgnacioCG, 2023-09-04:
Hi @jonathan1919810 I am working to develop a model in Pandaset, did you found a way to evaluate Pandaset?

Mingqj, 2023-10-28:
@1uk4s Hi, could you share your evaluation code on Pandaset? Thanks!