open-mmlab/OpenPCDet #1104: [Problem] Training with custom data reaches maximum recursion

leihui6, 2022-09-13:
Hi,

My environment is as follows:
```
torch              1.8.2+cu111
torchaudio         0.8.2
torchvision        0.9.2+cu111
RTX3090
Ubuntu 18.04
```
pcdet installed successfully, and demo runs perfectly.
However, when training my own dataset here is the error:

<details>
  <summary>error when training</summary>
  ### bash
python train.py --cfg_file cfgs/custom_models/pv_rcnn.yaml  --batch_size 64                      
2022-09-13 13:13:51,560   INFO  **********************Start logging**********************
2022-09-13 13:13:51,560   INFO  CUDA_VISIBLE_DEVICES=ALL
2022-09-13 13:13:51,560   INFO  cfg_file         cfgs/custom_models/pv_rcnn.yaml
2022-09-13 13:13:51,560   INFO  batch_size       64
2022-09-13 13:13:51,560   INFO  epochs           80
2022-09-13 13:13:51,560   INFO  workers          4
2022-09-13 13:13:51,560   INFO  extra_tag        default
2022-09-13 13:13:51,560   INFO  ckpt             None
2022-09-13 13:13:51,560   INFO  pretrained_model None
2022-09-13 13:13:51,560   INFO  launcher         none
2022-09-13 13:13:51,560   INFO  tcp_port         18888
2022-09-13 13:13:51,560   INFO  sync_bn          False
2022-09-13 13:13:51,560   INFO  fix_random_seed  False
2022-09-13 13:13:51,560   INFO  ckpt_save_interval 1
2022-09-13 13:13:51,561   INFO  local_rank       0
2022-09-13 13:13:51,561   INFO  max_ckpt_save_num 30
2022-09-13 13:13:51,561   INFO  merge_all_iters_to_one_epoch False
2022-09-13 13:13:51,561   INFO  set_cfgs         None
2022-09-13 13:13:51,561   INFO  max_waiting_mins 0
2022-09-13 13:13:51,561   INFO  start_epoch      0
2022-09-13 13:13:51,561   INFO  num_epochs_to_eval 0
2022-09-13 13:13:51,561   INFO  save_to_file     False
2022-09-13 13:13:51,561   INFO  use_tqdm_to_record False
2022-09-13 13:13:51,561   INFO  logger_iter_interval 50
2022-09-13 13:13:51,561   INFO  ckpt_save_time_interval 300
2022-09-13 13:13:51,561   INFO  wo_gpu_stat      False
2022-09-13 13:13:51,561   INFO  cfg.ROOT_DIR: /home/threed-detection/Desktop/OpenPCDet/OpenPCDet2/OpenPCDet
2022-09-13 13:13:51,561   INFO  cfg.LOCAL_RANK: 0
2022-09-13 13:13:51,561   INFO  cfg.CLASS_NAMES: ['Vehicle']
2022-09-13 13:13:51,561   INFO  
cfg.DATA_CONFIG = edict()
2022-09-13 13:13:51,561   INFO  cfg.DATA_CONFIG.DATASET: CustomDataset
2022-09-13 13:13:51,561   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/custom
2022-09-13 13:13:51,561   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [-4.8, -4.8, -2, 4.8, 4.8, 4]
2022-09-13 13:13:51,561   INFO  
cfg.DATA_CONFIG.MAP_CLASS_TO_KITTI = edict()
2022-09-13 13:13:51,561   INFO  cfg.DATA_CONFIG.MAP_CLASS_TO_KITTI.Vehicle: Car
2022-09-13 13:13:51,561   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2022-09-13 13:13:51,561   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2022-09-13 13:13:51,561   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2022-09-13 13:13:51,561   INFO  
cfg.DATA_CONFIG.INFO_PATH = edict()
2022-09-13 13:13:51,561   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['custom_infos_train.pkl']
2022-09-13 13:13:51,561   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['custom_infos_val.pkl']
2022-09-13 13:13:51,561   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2022-09-13 13:13:51,561   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2022-09-13 13:13:51,561   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2022-09-13 13:13:51,561   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2022-09-13 13:13:51,561   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2022-09-13 13:13:51,561   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
[... 489 lines cut ...]
    data_dict = cur_augmentor(data_dict=data_dict)
  File "../pcdet/datasets/augmentor/data_augmentor.py", line 49, in random_world_flip
    gt_boxes, points, enable = getattr(augmentor_utils, 'random_flip_along_%s' % cur_axis)(
  File "../pcdet/datasets/augmentor/augmentor_utils.py", line 16, in random_flip_along_x
    enable = np.random.choice([False, True], replace=False, p=[0.5, 0.5])
  File "mtrand.pyx", line 971, in numpy.random.mtrand.RandomState.choice
  File "<__array_function__ internals>", line 180, in count_nonzero
  File "/home/threed-detection/anaconda3/envs/openpcdet_py385/lib/python3.8/site-packages/numpy/core/numeric.py", line 492, in count_nonzero
    return multiarray.count_nonzero(a)
RecursionError: maximum recursion depth exceeded while calling a Python object
  ```
</details>
Sorry for the error log is too long, the command I used for training is:

```pyrhon
python -m pcdet.datasets.custom.custom_dataset create_custom_infos tools/cfgs/dataset_configs/custom_dataset.yaml 
python train.py --cfg_file cfgs/custom_models/pv_rcnn.yaml  --bach_size 64
```

Anyone has ideas what is going on? btw, I set the maximum recursion by `sys.setrecursionlimit(99999)` but it shows `RuntimeError: DataLoader worker (pid 13400) is killed by signal: Segmentation fault.` after long time no-response.

@jihanyang @OrangeSodahub 
I also found there was the same situation in [issure#1](https://github.com/open-mmlab/OpenPCDet/issues/953) [issure#2](https://github.com/open-mmlab/OpenPCDet/issues/1055)

Thanks in advance!

leihui6, 2022-09-13:
@jihanyang 
Thanks for your reply. I see, I checked the gt_boxes in `custom_dataset.py` as below
```
        if 'annos' in info:
            annos = info['annos']
            annos = common_utils.drop_info_with_name(annos, name='DontCare')
            gt_names = annos['name']
            gt_boxes_lidar = annos['gt_boxes_lidar']
            
            print(f"###########{index}")
            print(annos)
            print(gt_names)
            print(gt_boxes_lidar)
            print("###########")
            #exit()

            input_dict.update({
                'gt_names': gt_names,
                'gt_boxes': gt_boxes_lidar
            })
```
and the result is 
```
...
{'name': array(['Vehicle'], dtype='<U7'), 'gt_boxes_lidar': array([[ 9.19, -0.59, -0.59,  3.09,  1.46,  2.54,  2.08]], dtype=float32)}
['Vehicle']
[[ 9.19 -0.59 -0.59  3.09  1.46  2.54  2.08]]
###########
###########16
{'name': array(['Vehicle'], dtype='<U7'), 'gt_boxes_lidar': array([[ 9.19, -0.59, -0.59,  3.09,  1.46,  2.54,  2.08]], dtype=float32)}
['Vehicle']
[[ 9.19 -0.59 -0.59  3.09  1.46  2.54  2.08]]
###########
###########8
epochs:   0%|                                                                                   | 0/80 [00:01<?, ?it/s]
Traceback (most recent call last):
  File "train.py", line 221, in <module>
    main()
  File "train.py", line 168, in main
    train_model(
  File "/home/threed-detection/Desktop/OpenPCDet/OpenPCDet2/OpenPCDet/tools/train_utils/train_utils.py", line 150, in train_model
    accumulated_iter = train_one_epoch(
  File "/home/threed-detection/Desktop/OpenPCDet/OpenPCDet2/OpenPCDet/tools/train_utils/train_utils.py", line 30, in train_one_epoch
    batch = next(dataloader_iter)
  File "/home/threed-detection/anaconda3/envs/openpcdet_py385/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 517, in __next__
    data = self._next_data()
  File "/home/threed-detection/anaconda3/envs/openpcdet_py385/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1199, in _next_data
    return self._process_data(data)
  File "/home/threed-detection/anaconda3/envs/openpcdet_py385/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1225, in _process_data
    data.reraise()
  File "/home/threed-detection/anaconda3/envs/openpcdet_py385/lib/python3.8/site-packages/torch/_utils.py", line 429, in reraise
    raise self.exc_type(msg)
RecursionError: Caught RecursionError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/threed-detection/anaconda3/envs/openpcdet_py385/lib/python3.8/site-packages/torch/utils/data/_utils/worker.py", line 202, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/threed-detection/anaconda3/envs/openpcdet_py385/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/threed-detection/anaconda3/envs/openpcdet_py385/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "../pcdet/datasets/custom/custom_dataset.py", line 120, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "../pcdet/datasets/dataset.py", line 188, in prepare_data
    return self.__getitem__(new_index)

**so many __getiteam error**

```
I think the dataloader has no problem, what do you think?

jihanyang, 2022-09-14:
No. If you check line 188 in dataset.py, you will find that the error casued by there are no any gt_boxes in all scenes. I think you can set a checkpoint before the line 186. I suspect it is caused by this line 156:
https://github.com/open-mmlab/OpenPCDet/blob/bd96d39af2389478820b34db7ad0272e9bb205db/pcdet/datasets/dataset.py#L156
or line 170-171
https://github.com/open-mmlab/OpenPCDet/blob/bd96d39af2389478820b34db7ad0272e9bb205db/pcdet/datasets/dataset.py#L170-L171

leihui6, 2022-09-14:
Thanks again! @jihanyang 

I checked the go_boxes_mask and data_dict['gt_boxes'], all variable are not `None`.
Here is the interesting point I found, I add check the data_dict before and after line 182:

https://github.com/open-mmlab/OpenPCDet/blob/bd96d39af2389478820b34db7ad0272e9bb205db/pcdet/datasets/dataset.py#L182

in my code
``` python
print("before data_processor ###########")
print("data_dict['gt_boxes']->",data_dict['gt_boxes'])
data_dict = self.data_processor.forward(
    data_dict=data_dict
)
print("after data_processor ########")   
print("data_dict['gt_boxes']->",data_dict['gt_boxes'])
```

The result is always `None` after this function as below
```bash
data_dict['gt_boxes']-> [[ 7.3673673  -5.6462927  -0.57821673  3.1345434   2.109984    2.698345
  -1.9615599   1.        ]]
data_dict['gt_boxes']-> [[ 7.6473007 -4.48231   -0.5679081  2.9742985  1.4053321  2.4448926
   1.6139538  1.       ]]
after data_processor ########
after data_processor ########
data_dict['gt_boxes']-> []
data_dict['gt_boxes']-> []
before data_processor ###########
data_dict['gt_boxes']-> [[ 8.786528   -1.7666688  -0.39309707  3.0366747   1.5822157   2.4961662
  -2.2125812   1.        ]]
before data_processor ###########
data_dict['gt_boxes']-> [[ 7.325734    0.0397209  -0.47262332  3.1747956   2.1370792   2.7329957
  -1.8075655   1.        ]]
after data_processor ########
data_dict['gt_boxes']-> []
after data_processor ########
data_dict['gt_boxes']-> []
before data_processor ###########
before data_processor ###########
data_dict['gt_boxes']-> [[-9.800245  -2.356237  -0.5687006  3.1950634  2.1507223  2.750443
  -2.3673973  1.       ]]
data_dict['gt_boxes']-> [[-7.090519   -2.0712912  -0.48674133  3.0080614   2.024844    2.589464
  -1.3435211   1.        ]]
```
Why is that? should I modify something in `forward` function or check my dataset again?

jihanyang, 2022-09-14:
can you step into the data_processor and check which function cause that?

leihui6, 2022-09-14:
```bash
[[ 8.110613   -0.43972045 -0.533743    3.1716654   2.134972    2.7303011
  -1.6260003   1.        ]]
[]
[]
[]
[]

```

code: 
```python
    def forward(self, data_dict):
        """
        Args:
            data_dict:
                points: (N, 3 + C_in)
                gt_boxes: optional, (N, 7 + C) [x, y, z, dx, dy, dz, heading, ...]
                gt_names: optional, (N), string
                ...

        Returns:
        """

        for cur_processor in self.data_processor_queue:
            print(data_dict['gt_boxes'])
            data_dict = cur_processor(data_dict=data_dict)
            print(data_dict['gt_boxes'])

        return data_dict
```

jihanyang, 2022-09-15:
You should also `print(cur_processor)` and just step into it. It's obvious caused by the first processor and check what is the main problem. As the issue is not caused by the current codebase but your config or dataset prepare, I will close this issue.

closed 2022-09-15 by jihanyang

leihui6, 2022-09-15:
Thanks for your insight thoughts on the bug, I look through the `cur_processor` function list and found the `range of point cloud` was set unreasonably.  It can run smoothly right now.
Thank you again! 👍 @jihanyang