open-mmlab/OpenPCDet #1105: Seems that ‘--fix_random_seed’ doesn't work on ‘Second.yaml’ and train is unstable?

littleship123, 2022-09-14:
1.
I tried to fix random seeds for comparison. I found ‘--fix_random_seed’ doesn't work on 'Second.yaml', but 'PointPillar' seems correctly trained under fixed random seed. In fact, I found that there is also randomness on ‘Second_iou.yam’ (not in the figure).

I trained them on 8 GPUs and 80 EPOCHS on OpenPCDet of 0.6, and the training data has also been regenerated. Basically, I used the training settings provided by default.

However, I don't know which factor in Second model is causing this problem. What else might cause the training result to be not fixed？

2.
Besides, you can find that in the category of Pedestrian and Cyclist, the train results of Second are not stable enough. Does anyone know the reason? Is this a shortcoming of Second model?
[image]

jihanyang, 2022-09-14:
We have similar observation. I suspect this is caused by the implementation of Spconv, as it has many C++ implementation and the random seed in c++ cannot be fixed by python interface. Nevertheless, this hasn't be verified, just my understanding.

jihanyang, 2022-09-14:
The KITTI training set and validation set are both small, so the results of ped and cyclist are not stable. This is a common observation in KITTI.

littleship123, 2022-09-14:
> The KITTI training set and validation set are both small, so the results of ped and cyclist are not stable. This is a common observation in KITTI.

Thanks a lot. Could you tell me why the the results of Car is relatively stable? Did you mean “training set and validation set are both small" only for Pedestrian and Cyclists? Or is that the car category is also ”small“ but less affected by ”small“ factor?

jihanyang, 2022-09-15:
More samples.

closed 2022-09-15 by jihanyang

Sta8is, 2022-11-06:
> We have similar observation. I suspect this is caused by the implementation of Spconv, as it has many C++ implementation and the random seed in c++ cannot be fixed by python interface. Nevertheless, this hasn't be verified, just my understanding.

I am almost sure that spconv causes the problem. I managed to freeze the weights of 3D Backbone in SECOND and then fix_random_seed seems to be working (results are reproducible and same every time). However I dont know how to fix it? Maybe we must crate an issue here https://github.com/traveller59/spconv