open-mmlab/OpenPCDet #1107: Question: MPPnet

d33dler, 2022-09-14:
What is the inference time for MPPnet?

Cedarch, 2022-09-15:
The inference time (not including IO ) is about 200ms for 4-frame and 16frame's latency is about 50ms longer than 4frame ( resuing feature stored in a memory bank and running on V100).

xpyqiubai, 2022-09-16:
About the performance of MPPNet, are you aligned with the paper. Is there anything else that needs to be changed.

Cedarch, 2022-09-16:
Following the guideline, this implementation can achieve similar performance to the paper. 
Note: You need to generate multi-frame pkl first, and then train centerpoint_4frame as instructed. If the results of centerpoint_4frame are similar to the paper, you can get the reported performance of MPPNet，employing centerpoint_4frame as RPN.

xpyqiubai, 2022-09-16:
> Following the guideline, this implementation can achieve similar performance to the paper. Note: You need to generate multi-frame pkl first, and then train centerpoint_4frame as instructed. If the [...]

I have been based on the guideline to generate multi-frame pkl first. The details are shown in link [https://github.com/open-mmlab/OpenPCDet/issues/1111](url). But I could not get the results of centerpoint_4frame are similar to the paper. How do I align the first phase of MPPNet performance.

sshaoshuai, 2022-09-19:
Close this since it has been discussed in https://github.com/open-mmlab/OpenPCDet/issues/1111.

closed 2022-09-19 by sshaoshuai