open-mmlab/OpenPCDet #1109: SST support (Embracing Single Stride 3D Object Detector with Sparse Transformer)

jlqzzz, 2022-09-15:
SST is an excellent 3D detection method based on transformer, but at present there is only implementation code based on mmdetection3d. (https://github.com/tusen-ai/SST)

Can you provide SST support based on OpenPCDet?

sshaoshuai, 2022-09-16:
It should be easy to support SST in OpenPCDet, and the most important thing you need to pay attention is to check the coordinate system between these two repos. Currently we do not have much time to do experiments with SST, maybe we will consider to support it in Nov. or Dec. this year.

jlqzzz, 2022-09-17:
> It should be easy to support SST in OpenPCDet, and the most important thing you need to pay attention is to check the coordinate system between these two repos. Currently we do not have much time to do experiments with SST, maybe we will consider to support it in Nov. or Dec. this year.

Thanks! 
Look forward to it!

closed 2022-11-02 by github-actions[bot]