open-mmlab/OpenPCDet #1116: Submit on KITTI website

JLIN77, 2022-09-20:
Yes. I did inference the KITTI test set with official PV-RCNN weight, and submit the results to KITTI website. The results show as:
Car (3D Detection) | 66.01 % | 64.26 % | 61.65 %
Pedestrian (3D Detection) | 35.58 % | 29.18 % | 27.36 %
Cyclist (3D Detection) | 51.01 % | 42.79 % | 39.31 %
Compared with mmlab-PVRCNN model on KITTI leaderboard, it drops almost 10~20 mAP.
Any answer to this question?

sshaoshuai, 2022-09-26:
It seems strange, can you fisrt evaluate the performance on KITTI validation set? Generally the performance between test and validation will not be very large like that.

MLbeginner2, 2022-10-24:
I met the same problem.I did inference the KITTI test set with official PointPillar_7728.pth, and submit the results to KITTI website. The results show as:
Car (3D Detection) | 82.31 % | 72.98 % | 66.73 %
Pedestrian (3D Detection) | 40.65 % | 32.74 % | 30.19 %
Cyclist (3D Detection) | 69.39 % | 54.12 % | 47.82 %
 This is too strange, could it be that the pcdet is not sufficiently trained?

MLbeginner2, 2022-10-24:
> I met the same problem.I did inference the KITTI test set with official PointPillar_7728.pth, and submit the results to KITTI website. The results show as: Car (3D Detection) | 82.31 % | 72.98 % | [...]
I am training on all KITTI data with labels, not 3712.

closed 2022-12-08 by github-actions[bot]

BravoCharlie-ai, 2023-02-11:
Hello guys,

Can you let me know how you convert pickle file to submit on leaderboard?
Do I need to do the coordinate transformation? I uploaded the result but I got zero for 3D detection. Whereas Bbbox results are sufficiently high?

MLbeginner2, 2023-02-13:
> Hello guys,
> 
> Can you let me know how you convert pickle file to submit on leaderboard? Do I need to do the coordinate transformation? I uploaded the result but I got zero for 3D detection. Whereas Bbbox results are sufficiently high?

run python test.py ********** --save_to_file

BravoCharlie-ai, 2023-02-13:
Thank you for the reply. But I still face the same issue. I even don't see 3D detections in the leaderboad result.

![Screenshot from 2023-02-13 08-37-17]
(https://user-images.githubusercontent.com/52042236/218472695-1c64d215-74b7-4828-a0b7-1ced3e3f85b4.png)

BravoCharlie-ai, 2023-02-13:
Thank you, My issue is resolved actually.

huangmang, 2023-03-05:
> Thank you, My issue is resolved actually.

hello I met the same problem .I don't see 3d detection result.Cou you tell me how to solve this problem,thanks a lot.

huangmang, 2023-03-06:
> 
hello, my test result dosen't have 3d ,have you met this problem ？

huangmang, 2023-03-06:
> Yes. I did inference the KITTI test set with official PV-RCNN weight, and submit the results to KITTI website. The results show as: Car (3D Detection) | 66.01 % | 64.26 % | 61.65 % Pedestrian (3D [...]

hello, my test result dosen't have 3d ,have you met this problem ？
[image]

NiranjanRavi1993, 2023-09-26:
Hi all, I am facing the same issue. Pre-trained model/official reported results from pcdet repos are dropping by 1-3 mAP when submission to Kitti test set evaluation. Anyone was able to find a reason/solution?

Petros626, 2024-10-24:
> aakashjuseja
help this user with your solution @huangmang

Petros626, 2024-10-24:
@aakashjuseja-aj help him @huangmang