open-mmlab/OpenPCDet #1117: The results in paper are different to the Leaderboard of waymo dataset challenge website.

sjtuljw520, 2022-09-21:
I find that the results in Table 3 of paper ("MPPNet: Multi-Frame Feature Intertwining with Proxy Points for 3D Temporal Object Detection") is different to the  Leaderboard in waymo website. The leaderboard results are better. Can you please provide the way to realize the results on the leaderboard?

Cedarch, 2022-09-21:
Hi, the reported performance in table 3  of our paper is results without TTA (test-time-augment) and model ensemble and there is about 3~4% gain in terms of mAPH when using TTA and model ensmble (named MPPNetEns).  if you have achieved similar results with our paper on val dataset, you can follow the the instruction from [https://arxiv.org/pdf/2112.09205.pdf](url) to conduct model ensemble and TTA. The performance of ensembled results are sensitive to hyper-parameters, so you may need to spend some time finding the best parameters for your models

sjtuljw520, 2022-09-21:
OK, many thanks for your suggestion. @Cedarch

closed 2022-09-26 by sshaoshuai

hdddhdd, 2024-10-31:
Hello,

Thank you for the great work on this project. I'm interested in exploring model ensemble techniques and would like to try them with this repository. Could you please share a code example or guidance on implementing model ensemble with the current setup? Any examples or documentation would be really helpful!

Thanks in advance for your help!