open-mmlab/OpenPCDet #1120: Provide more details about the evaluation results?

sjtuljw520, 2022-09-22:
The results provided by the evaluation function do not contain metrics in different distance. Can you improve the function which can provide results like follows:

OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1: [mAP 0.657237] [mAPH 0.652432]
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2: [mAP 0.579091] [mAPH 0.574783]
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1: [mAP 0.751253] [mAPH 0.708188]
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2: [mAP 0.676412] [mAPH 0.636359]
OBJECT_TYPE_TYPE_SIGN_LEVEL_1: [mAP 0] [mAPH 0]
OBJECT_TYPE_TYPE_SIGN_LEVEL_2: [mAP 0] [mAPH 0]
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1: [mAP 0.63183] [mAPH 0.621806]
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2: [mAP 0.609326] [mAPH 0.599654]
RANGE_TYPE_VEHICLE_[0, 30)_LEVEL_1: [mAP 0.860443] [mAPH 0.855564]
RANGE_TYPE_VEHICLE_[0, 30)_LEVEL_2: [mAP 0.847752] [mAPH 0.842935]
RANGE_TYPE_VEHICLE_[30, 50)_LEVEL_1: [mAP 0.623048] [mAPH 0.618002]
RANGE_TYPE_VEHICLE_[30, 50)_LEVEL_2: [mAP 0.564474] [mAPH 0.55984]
RANGE_TYPE_VEHICLE_[50, +inf)_LEVEL_1: [mAP 0.386604] [mAPH 0.381241]
RANGE_TYPE_VEHICLE_[50, +inf)_LEVEL_2: [mAP 0.293866] [mAPH 0.289682]
RANGE_TYPE_PEDESTRIAN_[0, 30)_LEVEL_1: [mAP 0.796615] [mAPH 0.752106]
RANGE_TYPE_PEDESTRIAN_[0, 30)_LEVEL_2: [mAP 0.758854] [mAPH 0.715435]
RANGE_TYPE_PEDESTRIAN_[30, 50)_LEVEL_1: [mAP 0.739176] [mAPH 0.697473]
RANGE_TYPE_PEDESTRIAN_[30, 50)_LEVEL_2: [mAP 0.673429] [mAPH 0.63439]
RANGE_TYPE_PEDESTRIAN_[50, +inf)_LEVEL_1: [mAP 0.659436] [mAPH 0.620717]
RANGE_TYPE_PEDESTRIAN_[50, +inf)_LEVEL_2: [mAP 0.532685] [mAPH 0.499568]
RANGE_TYPE_SIGN_[0, 30)_LEVEL_1: [mAP 0] [mAPH 0]
RANGE_TYPE_SIGN_[0, 30)_LEVEL_2: [mAP 0] [mAPH 0]
RANGE_TYPE_SIGN_[30, 50)_LEVEL_1: [mAP 0] [mAPH 0]
RANGE_TYPE_SIGN_[30, 50)_LEVEL_2: [mAP 0] [mAPH 0]
RANGE_TYPE_SIGN_[50, +inf)_LEVEL_1: [mAP 0] [mAPH 0]
RANGE_TYPE_SIGN_[50, +inf)_LEVEL_2: [mAP 0] [mAPH 0]
RANGE_TYPE_CYCLIST_[0, 30)_LEVEL_1: [mAP 0.705526] [mAPH 0.694948]
RANGE_TYPE_CYCLIST_[0, 30)_LEVEL_2: [mAP 0.700466] [mAPH 0.689964]
RANGE_TYPE_CYCLIST_[30, 50)_LEVEL_1: [mAP 0.596796] [mAPH 0.5874]
RANGE_TYPE_CYCLIST_[30, 50)_LEVEL_2: [mAP 0.565472] [mAPH 0.556557]
RANGE_TYPE_CYCLIST_[50, +inf)_LEVEL_1: [mAP 0.498838] [mAPH 0.489147]
RANGE_TYPE_CYCLIST_[50, +inf)_LEVEL_2: [mAP 0.465546] [mAPH 0.456499]

sshaoshuai, 2022-09-26:
We do not provide this by default since it causes too much calculations. 
You can support his by modifying here: https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/waymo/waymo_eval.py#L93-L106.

For detailed guidance please refer to the official repo of Waymo Open Dataset.

sjtuljw520, 2022-09-27:
OK, thank you for your help. @sshaoshuai

closed 2022-09-27 by sshaoshuai