open-mmlab/OpenPCDet #1122: training on multiple gpu

suvasis, 2022-09-23:
hi,

I have 2 gpu machine. Single GPU training works ok. However for 2 gpus, It seems to be waiting for something for ever. The GPU utilization is 0% for both GPUs.

I am using Kitti dataset as described in the documentation. 

The command as shown: 

OpenPCDet/tools$ ./scripts/dist_train.sh 2 --batch_size 2 --epochs 1 --cfg_file  cfgs/kitti_models/pv_rcnn.yaml

My code base is as of Sep 22, 2022. 

Log snippet:

2022-09-23 10:21:21,551   INFO  **********************Start training kitti_models/pv_rcnn(default)**********************
epochs: 0it [00:07, ?it/s]
2022-09-23 10:21:30,211   INFO  **********************End training kitti_models/pv_rcnn(default)**********************

2022-09-23 10:21:30,212   INFO  **********************Start evaluation kitti_models/pv_rcnn(default)**********************
2022-09-23 10:21:30,213   INFO  Loading KITTI dataset
2022-09-23 10:21:30,304   INFO  Total samples for KITTI dataset: 3769
Wait 30 seconds for next check (progress: 0.0 / 0 minutes): /home/minasm/suvasis/tools/pvrcnn/OpenPCDet/output/kitti_models/pv_rcnn/default/ckpt

LOG SNIPPET:

 NGPUS=2
+ PY_ARGS='--batch_size 2 --epochs 1 --cfg_file cfgs/kitti_models/pv_rcnn.yaml'
+ true
+ PORT=27915
++ nc -z 127.0.0.1 27915
++ echo 1
+ status=1
+ '[' 1 '!=' 0 ']'
+ break
+ echo 27915
27915
+ echo --batch_size 2 --epochs 1 --cfg_file cfgs/kitti_models/pv_rcnn.yaml
--batch_size 2 --epochs 1 --cfg_file cfgs/kitti_models/pv_rcnn.yaml
+ python -m torch.distributed.launch --nproc_per_node=2 --rdzv_endpoint=localhost:27915 train.py --launcher pytorch --batch_size 2 --epochs 1 --cfg_file cfgs/kitti_models/pv_rcnn.yaml
/home/minasm/suvasis/tools/anaconda3/envs/pytorchbuild/lib/python3.9/site-packages/torch/distributed/launch.py:178: FutureWarning: The module torch.distributed.launch is deprecated
and will be removed in future. Use torchrun.
Note that --use_env is set by default in torchrun.
If your script expects `--local_rank` argument to be set, please
change it to read from `os.environ['LOCAL_RANK']` instead. See
https://pytorch.org/docs/stable/distributed.html#launch-utility for
further instructions

  warnings.warn(
WARNING:torch.distributed.run:
*****************************************
Setting OMP_NUM_THREADS environment variable for each process to be 1 in default, to avoid your system being overloaded, please further tune the variable for optimal performance in your application as needed.
*****************************************
2022-09-23 10:21:20,169   INFO  **********************Start logging**********************
2022-09-23 10:21:20,169   INFO  CUDA_VISIBLE_DEVICES=ALL
2022-09-23 10:21:20,169   INFO  total_batch_size: 2
2022-09-23 10:21:20,169   INFO  cfg_file         cfgs/kitti_models/pv_rcnn.yaml
2022-09-23 10:21:20,169   INFO  batch_size       1
2022-09-23 10:21:20,170   INFO  epochs           1
2022-09-23 10:21:20,170   INFO  workers          4
2022-09-23 10:21:20,170   INFO  extra_tag        default
2022-09-23 10:21:20,170   INFO  ckpt             None
2022-09-23 10:21:20,170   INFO  pretrained_model None
2022-09-23 10:21:20,170   INFO  launcher         pytorch
2022-09-23 10:21:20,170   INFO  tcp_port         18888
2022-09-23 10:21:20,170   INFO  sync_bn          False
2022-09-23 10:21:20,170   INFO  fix_random_seed  False
2022-09-23 10:21:20,170   INFO  ckpt_save_interval 1
2022-09-23 10:21:20,170   INFO  local_rank       0
2022-09-23 10:21:20,170   INFO  max_ckpt_save_num 30
2022-09-23 10:21:20,170   INFO  merge_all_iters_to_one_epoch False
[... 564 lines cut ...]
        (5): BatchNorm1d(256, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
        (6): ReLU()
        (7): Conv1d(256, 1, kernel_size=(1,), stride=(1,))
      )
      (reg_layers): Sequential(
        (0): Conv1d(256, 256, kernel_size=(1,), stride=(1,), bias=False)
        (1): BatchNorm1d(256, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
        (2): ReLU()
        (3): Dropout(p=0.3, inplace=False)
        (4): Conv1d(256, 256, kernel_size=(1,), stride=(1,), bias=False)
        (5): BatchNorm1d(256, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
        (6): ReLU()
        (7): Conv1d(256, 7, kernel_size=(1,), stride=(1,))
      )
    )
  )
)
2022-09-23 10:21:21,551   INFO  **********************Start training kitti_models/pv_rcnn(default)**********************
epochs: 0it [00:07, ?it/s]
2022-09-23 10:21:30,211   INFO  **********************End training kitti_models/pv_rcnn(default)**********************

2022-09-23 10:21:30,212   INFO  **********************Start evaluation kitti_models/pv_rcnn(default)**********************
2022-09-23 10:21:30,213   INFO  Loading KITTI dataset
2022-09-23 10:21:30,304   INFO  Total samples for KITTI dataset: 3769
Wait 30 seconds for next check (progress: 0.0 / 0 minutes): /home/minasm/suvasis/tools/pvrcnn/OpenPCDet/output/kitti_models/pv_rcnn/defaulWait 30 seconds for next check (progress: 0.5 / 0 minutes): /home/minasm/suvasis/tools/pvrcnn/OpenPCDet/output/kitti_models/pv_rcnn/defaulWait 30 seconds for next check (progress: 1.0 / 0 minutes): /home/minasm/suvasis/tools/pvrcnn/OpenPCDet/output/kitti_models/pv_rcnn/defaulWait 30 seconds for next check (progress: 1.5 / 0 minutes): /home/minasm/suvasis/tools/pvrcnn/OpenPCDet/output/kitti_models/pv_rcnn/defaul

jihanyang, 2022-09-24:
```
2022-09-23 10:21:21,551 INFO Start training kitti_models/pv_rcnn(default)
epochs: 0it [00:07, ?it/s]
2022-09-23 10:21:30,211 INFO End training kitti_models/pv_rcnn(default)
```
As shown in your log, the training has been finished in previous lauch.

suvasis, 2022-09-24:
hi,

After I followed this comment https://github.com/open-mmlab/OpenPCDet/issues/938
    #Have you tried to comment these two lines to quickly start the training?
    #if mp.get_start_method(allow_none=True) is None:
    #    mp.set_start_method('spawn')

The training worked only for batch_size 2. If I increase the batch_size  to 4 or beyond (the log snippet is attached below). What should I do to fix this?

For batch_size=2, the run the successful. For batch_size=4, the training fails.

///////////////////////////////////////////////////////////////////////////////////////////

batch_size=2

///////////////////////////////////////////////////////////////////////////////////////////

command:
(pytorchbuild) minasm@lambda-quad:~/suvasis/tools/pvrcnn/OpenPCDet/tools$ ./scripts/dist_train.sh 2 --batch_size 2 --epochs 10 --cfg_file  cfgs/kitti_models/pv_rcnn.yaml

log:
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:90.7043, 78.4946, 74.1905
bev  AP:90.0927, 74.9485, 70.5419
3d   AP:90.0911, 74.9443, 70.5419
aos  AP:90.47, 76.53, 72.11
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:94.0771, 79.2091, 75.3161
bev  AP:93.6473, 75.5676, 72.0880
3d   AP:93.6469, 75.5664, 72.0693
aos  AP:93.83, 77.14, 73.01

2022-09-23 14:07:09,004   INFO  Result is save to /home/minasm/suvasis/tools/pvrcnn/OpenPCDet/output/kitti_models/pv_rcnn/default/eval/eval_with_train/epoch_10/val
2022-09-23 14:07:09,005   INFO  ****************Evaluation done.*****************
2022-09-23 14:07:09,026   INFO  Epoch 10 has been evaluated
Wait 30 seconds for next check (progress: 0.0 / 0 minutes): /home/minasm/suvasis/tools/pvrcnn/OpenPCDet/output/kitti_models/pv_rcnn/defaul2022-09-23 14:07:39,058   INFO  **********************End evaluation kitti_models/pv_rcnn(default)**********************
///////////////////////////////////////////////////////////////////////////////////////////

batch_size=4

///////////////////////////////////////////////////////////////////////////////////////////

command:
OpenPCDet/tools$ ./scripts/dist_train.sh 2 --batch_size 4 --epochs 10 --cfg_file  cfgs/kitti_models/pv_rcnn.yaml

Log:

pochs:   0%|                                                                                                      | 0/10 [00:00<?, ?it/s]2022-09-23 17:46:02,631   INFO  **********************Start training kitti_models/pv_rcnn(default)**********************
epochs:   0%|                                                                                                      | 0/10 [00:00<?, ?it/s/home/minasm/suvasis/tools/anaconda3/envs/pytorchbuild/lib/python3.9/site-packages/torch/autograd/__init__.py:173: UserWarning: Grad strides do not match bucket view strides. This may indicate grad was not created according to the gradient layout contract, or that the param's strides changed since DDP was constructed.  This is not an error, but may impair performance.
grad.sizes() = [7, 256, 1], strides() = [256, 1, 256]
bucket_view.sizes() = [7, 256, 1], strides() = [256, 1, 1] (Triggered internally at  /opt/conda/conda-bld/pytorch_1659484806139/work/torch/csrc/distributed/c10d/reducer.cpp:312.)
  Variable._execution_engine.run_backward(  # Calls into the C++ engine to run the backward pass
/home/minasm/suvasis/tools/anaconda3/envs/pytorchbuild/lib/python3.9/site-packages/torch/autograd/__init__.py:173: UserWarning: Grad strides do not match bucket view strides. This may indicate grad was not created according to the gradient layout contract, or that the param's strides changed since DDP was constructed.  This is not an error, but may impair performance.
grad.sizes() = [7, 256, 1], strides() = [256, 1, 256]
bucket_view.sizes() = [7, 256, 1], strides() = [256, 1, 1] (Triggered internally at  /opt/conda/conda-bld/pytorch_1659484806139/work/torch/csrc/distributed/c10d/reducer.cpp:312.)
  Variable._execution_engine.run_backward(  # Calls into the C++ engine to run the backward pass
2022-09-23 17:46:09,141   INFO  epoch: 0/10, acc_iter=1, cur_iter=0/928, batch_size=2, time_cost(epoch): 00:06/1:35:24, time_cost(all): 00:06/15:54:01, loss=9.989889144897461, d_time=1.04(1.04), f_time=5.11(5.11), b_time=6.15(6.15), lr=0.0009999999999999992
Traceback (most recent call last):
  File "/home/minasm/suvasis/tools/pvrcnn/OpenPCDet/tools/train.py", line 221, in <module>
    main()
  File "/home/minasm/suvasis/tools/pvrcnn/OpenPCDet/tools/train.py", line 168, in main
    train_model(
  File "/home/minasm/suvasis/tools/pvrcnn/OpenPCDet/tools/train_utils/train_utils.py", line 150, in train_model
    accumulated_iter = train_one_epoch(
  File "/home/minasm/suvasis/tools/pvrcnn/OpenPCDet/tools/train_utils/train_utils.py", line 54, in train_one_epoch
    loss.backward()
  File "/home/minasm/suvasis/tools/anaconda3/envs/pytorchbuild/lib/python3.9/site-packages/torch/_tensor.py", line 396, in backward
    torch.autograd.backward(self, gradient, retain_graph, create_graph, inputs=inputs)
  File "/home/minasm/suvasis/tools/anaconda3/envs/pytorchbuild/lib/python3.9/site-packages/torch/autograd/__init__.py", line 173, in backward
    Variable._execution_engine.run_backward(  # Calls into the C++ engine to run the backward pass
[... 48 lines cut ...]
|   0  NVIDIA GeForce ...  On   | 00000000:01:00.0 Off |                  N/A |
| 30%   39C    P8    15W / 220W |     67MiB /  8192MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
[... cut ...]
|    0   N/A  N/A      1733      G   /usr/lib/xorg/Xorg                 56MiB |
|    0   N/A  N/A      1908      G   /usr/bin/gnome-shell                8MiB |
|    1   N/A  N/A      1733      G   /usr/lib/xorg/Xorg                  4MiB |
+-----------------------------------------------------------------------------+

tools$ nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2022 NVIDIA Corporation
Built on Wed_Jun__8_16:49:14_PDT_2022
Cuda compilation tools, release 11.7, V11.7.99
Build cuda_11.7.r11.7/compiler.31442593_0

closed 2022-10-24 by suvasis