open-mmlab/OpenPCDet #1124: compatible with torch1.12

OuyangJunyuan, 2022-09-25:
`"<THC/THC.h> no such file or directory"` occured when I try to build `OpenPCDet` with `torch1.12`. 
I solve this problem as following:
1. remove all `<THC/THC.h>` in `ops` dir.
2. replace all `xxx.data<type>()` in `ops` dir with `xxx.data_ptr<type>()`.
3. replace all `x.type().is_cuda()` in `ops` dir with `x.is_cuda()`.
4. remove all `extern THCstate state` in `ops` dir.

sshaoshuai, 2022-09-26:
We have solved a similar issue here: https://github.com/open-mmlab/OpenPCDet/pull/1041. 
Can you look into this PR and check whether it solves your issue?

OuyangJunyuan, 2022-09-28:
> We have solved a similar issue here: #1041. Can you look into this PR and check whether it solves your issue?

thanks. It works for me

closed 2022-11-13 by github-actions[bot]