open-mmlab/OpenPCDet #1129: The evaluation results are bad for PV-RCNN++/PV-RCNN++ resnet

sjtuljw520, 2022-09-27:
The evaluation results as follows:

**PV-RCNN ++ (trained with 20% waymo training data):**
2022-09-25 13:47:13,688   INFO  *************** Performance of EPOCH 30 *****************
2022-09-25 13:47:13,688   INFO  Generate label finished(sec_per_example: 0.0324 second).
2022-09-25 13:47:13,689   INFO  recall_roi_0.3: 0.397708
2022-09-25 13:47:13,689   INFO  recall_rcnn_0.3: 0.398417
2022-09-25 13:47:13,689   INFO  recall_roi_0.5: 0.340692
2022-09-25 13:47:13,689   INFO  recall_rcnn_0.5: 0.348162
2022-09-25 13:47:13,689   INFO  recall_roi_0.7: 0.173923
2022-09-25 13:47:13,689   INFO  recall_rcnn_0.7: 0.198185
2022-09-25 13:47:15,227   INFO  Average predicted number of objects(39987 samples): 19.979
2022-09-25 13:54:56,163   INFO  
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.0647 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.0623 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.0554 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.0534 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.0978 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.0835 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.0813 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.0694 
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/AP: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/APH: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/AP: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/APH: 0.0000 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.5279 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.5170 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.5085 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.4980 

**PV-RCNN ++ with resnet (trained with 20% waymo training data):**
2022-09-25 16:07:03,950   INFO  *************** Performance of EPOCH 30 *****************
2022-09-25 16:07:03,951   INFO  Generate label finished(sec_per_example: 0.0992 second).
2022-09-25 16:07:03,951   INFO  recall_roi_0.3: 0.379494
2022-09-25 16:07:03,951   INFO  recall_rcnn_0.3: 0.379777
2022-09-25 16:07:03,951   INFO  recall_roi_0.5: 0.328294
2022-09-25 16:07:03,951   INFO  recall_rcnn_0.5: 0.333714
2022-09-25 16:07:03,951   INFO  recall_roi_0.7: 0.169521
2022-09-25 16:07:03,951   INFO  recall_rcnn_0.7: 0.189351
2022-09-25 16:07:05,201   INFO  Average predicted number of objects(39987 samples): 14.846
2022-09-25 16:15:17,837   INFO  
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.0460 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.0443 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.0394 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.0380 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.0736 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.0633 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.0611 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.0526 
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/AP: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/APH: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/AP: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/APH: 0.0000 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.4488 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.4403 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.4324 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.4243 

**PV-RCNN ++ with resnet (trained with 100% waymo training data):**
2022-09-22 17:32:16,046   INFO  *************** Performance of EPOCH 30 *****************
2022-09-22 17:32:16,046   INFO  Generate label finished(sec_per_example: 0.1513 second).
2022-09-22 17:32:16,046   INFO  recall_roi_0.3: 0.285793
2022-09-22 17:32:16,047   INFO  recall_rcnn_0.3: 0.286112
2022-09-22 17:32:16,047   INFO  recall_roi_0.5: 0.244632
2022-09-22 17:32:16,047   INFO  recall_rcnn_0.5: 0.248490
2022-09-22 17:32:16,047   INFO  recall_roi_0.7: 0.122974
2022-09-22 17:32:16,047   INFO  recall_rcnn_0.7: 0.135108
2022-09-22 17:32:21,374   INFO  Average predicted number of objects(39987 samples): 15.287
2022-09-22 17:43:38,581   INFO  
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.0435 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.0421 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.0372 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.0360 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.0627 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.0541 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.0520 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.0449 
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/AP: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/APH: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/AP: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/APH: 0.0000 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.3447 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.3372 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.3321 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.3249 

从结果来看，检测的召回率很低（result.pkl文件大小70Mb左右），而且都是VEHICLE和PEDESTRIAN接近为0。
现在的数据版本是：training dataset 是waymo dataset V1.0；validation dataset是waymo dataset V1.2;
不知道是因为算法默认的配置文件需要做调整？还是training dataset 数据版本的影响？

sshaoshuai, 2022-09-27:
Have you modified any parts of the codes? The performance is not correct and we suggest you follow the README.md to prepare your data and run the training & evaluation process. 
Currently the provided information is limited and we also can not know where the problem is. 
In my side, training with v1.0 and evaluating with v1.2 should be fine to get similar performance.

sjtuljw520, 2022-09-28:
Hi, I did not modify the codes, just modify the dataset dirs in the config file. @sshaoshuai

sshaoshuai, 2022-10-01:
Then maybe there are some inconsistencies in your data. I suggest you follow the guideline to re-prepare the data and re-train it, since we have trained it multiple times and the performance is normal.

sjtuljw520, 2022-10-27:
Thank you. I have re download the new version dataset and re-trained the pv-rcnn++(resnet) model. Now the evaluation results are close to the results you provide. @sshaoshuai

closed 2022-12-12 by github-actions[bot]