open-mmlab/OpenPCDet #1134: On evaluating metrics up on the validation set and down on the test set

MLbeginner2, 2022-10-01:
Hello everyone, I have done a set of experiments on KITTI using OpenPCDet framework. There is some improvement in the evaluation metrics on the validation set, but the metrics of cyclists and pedestrians on the test set have greatly decreased. I guess that might be overfitting, so I want to ask how do I avoid overfitting? Is it by reducing the number of training epochs? Also, are KITTI's validation set and test set positively correlated? If the metrics go up on the KITTI validation set, does the metrics necessarily go up on the test set?
In the Figure below, P represents Pointpillar, I  represents My_Net.
[image]

jihanyang, 2022-10-01:
```
Also, are KITTI's validation set and test set positively correlated? If the metrics go up on the KITTI validation set, does the metrics necessarily go up on the test set?
```
No. Actually, as the number of samples of KITTI is limited, the performance on cyclist and pedestrian fluctuate a lot on the validation set for the same model. If the performance of KITTI validation set and test set are not tightly correlated.

MLbeginner2, 2022-10-08:
> ```
> Also, are KITTI's validation set and test set positively correlated? If the metrics go up on the KITTI validation set, does the metrics necessarily go up on the test set?
> ```
>
> No. Actually, as [...]

For experiments using the KITTI dataset in a general paper, should I use all 7481 samples for training or only 3712 for training?

closed 2022-11-23 by github-actions[bot]

bqcs, 2024-01-14:
> the metrics of cyclists and pedestrians on the test set have greatly decreased

Hi，‘’The metrics of cyclists and pedestrians on the test set have greatly decreased‘’. 
I've also encountered this issue, do you have a solution?