open-mmlab/OpenPCDet #1140: Does kitti_object_eval_python/eval.py have a mistake?

dingluchuan, 2022-10-09:
Hi, I am using the eval.py to eval my data，but I find the line 340
def calculate_iou_partly(gt_annos, dt_annos, metric, num_parts=50)
however the line 473 when the function is using:
rets = calculate_iou_partly(dt_annos, gt_annos, metric, num_parts)
the dt_annos，gt_annos change the loction of the function，is it a mistake?
thanks

dingluchuan, 2022-10-09:
the calculate_iou_partly function return overlaps, parted_overlaps, total_gt_num, total_dt_num at line 414
and when using overlaps, parted_overlaps, total_gt_num, total_dt_num = rets at line 474
It is very confused.

shashankag14, 2022-10-09:
This might be helpful.
https://github.com/open-mmlab/OpenPCDet/pull/1085

dingluchuan, 2022-10-10:
yeah，thanks.

closed 2022-10-12 by jihanyang