open-mmlab/OpenPCDet #1141: about the version of package "kornia"

WYYAHYT, 2022-10-10:
I want to run CaDDN with pcdet，but failed with error:  **name 'normalize' is not defined**
```
Traceback (most recent call last):
  File "/home/chenhai-fwxz/OpenPCDet/tools/test.py", line 201, in <module>
    main()
  File "/home/chenhai-fwxz/OpenPCDet/tools/test.py", line 197, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "/home/chenhai-fwxz/OpenPCDet/tools/test.py", line 65, in eval_single_ckpt
    result_dir=eval_output_dir, save_to_file=args.save_to_file
  File "/home/chenhai-fwxz/OpenPCDet/tools/eval_utils/eval_utils.py", line 58, in eval_one_epoch
    pred_dicts, ret_dict = model(batch_dict) 
  File "/home/chenhai-fwxz/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/nn/modules/module.py", line 550, in __call__
    result = self.forward(*input, **kwargs)
  File "../pcdet/models/detectors/caddn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/chenhai-fwxz/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/nn/modules/module.py", line 550, in __call__
    result = self.forward(*input, **kwargs)
  File "../pcdet/models/backbones_3d/vfe/image_vfe.py", line 72, in forward
    batch_dict = self.ffn(batch_dict)
  File "/home/chenhai-fwxz/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/nn/modules/module.py", line 550, in __call__
    result = self.forward(*input, **kwargs)
  File "../pcdet/models/backbones_3d/vfe/image_vfe_modules/ffn/depth_ffn.py", line 51, in forward
    ddn_result = self.ddn(images)
  File "/home/chenhai-fwxz/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/nn/modules/module.py", line 550, in __call__
    result = self.forward(*input, **kwargs)
  File "../pcdet/models/backbones_3d/vfe/image_vfe_modules/ffn/ddn/ddn_template.py", line 121, in forward
    x = self.preprocess(images)
  File "../pcdet/models/backbones_3d/vfe/image_vfe_modules/ffn/ddn/ddn_template.py", line 158, in preprocess
    x = normalize(x, mean=self.norm_mean, std=self.norm_std)
NameError: name 'normalize' is not defined
```

veriosn of my env:
```
kornia                        0.3.0
torch                         1.5.0
python                      3.7.12
```
with cuda 10.2

**I try to change the version of kornia and change the version of torch and python corresponding to it, 
but encountered another error as shown below：**
`undefined symbol: _ZN 3c105ErrorC2ENS_14 SourceLocation lESs`

closed 2022-10-11 by WYYAHYT

Ysnnnn, 2024-01-24:
have you solved this problem? I met it too

Petros626, 2024-10-24:
usw this:
```txt
numpy
llvmlite
numba
torch>=1.1
tensorboardX
easydict
pyyaml
scikit-image
tqdm
torchvision
SharedArray
opencv-python
pyquaternion
av2==0.2.0
kornia==0.6.5
```