open-mmlab/OpenPCDet #1142: Very large RAM occupation for MPPNET.

sjtuljw520, 2022-10-11:
The RAM occupation seems very large for mppnet. When I set the num_of_workers as 2 and trained the model on 2 GPU (batch_size_per_gpu = 4), 480GiB RAM is work out, the program is killed as OOM. Is this normal? And, depend on your experience, how much RAM needed to train MPPNET.

Cedarch, 2022-10-12:
Hi, we train MPPNet with batch_size_per_gpu = 2 and 512G RAM is enough.  You can try this setting.

sjtuljw520, 2022-10-12:
Thanks for replying. I am doubt why the model cosume so much RAM. Does the dataloader load all the data at once, instead of batch by batch? @Cedarch

xpyqiubai, 2022-10-13:
I train MPPNet with batch_size_per_gpu = 2 and 240G RAM is enough.

sjtuljw520, 2022-10-13:
Thank you! What is the number of workers you set? I set batch_size_per_gpu =2, and num_of_workers = 1. However, 512G RAM is till not enough.  @xpyqiubai 
BTW, I trained the model in one GPU.

Cedarch, 2022-10-14:
The dataloader just load data by batch and each item contains 4-frame concated point clouds. It actually cost larger RAM than single frame setting but, for our experience, 512G RAM is enough, @sjtuljw520.

sjtuljw520, 2022-11-04:
I retrain the model with new version of waymo dataset (the dataset vesion I used before is v1.1). Now this problem has been solved. 120G RAM is enough. Thank you @Cedarch

closed 2022-12-20 by github-actions[bot]