open-mmlab/OpenPCDet #1143: the test result in pvrcnn could not reach the parper result

JianyyuWang, 2022-10-11:
i test the pvrcnn_8369 and the test result is 

Car (3D Detection) | 87.45 % | 80.25 % | 76.19 %
Pedestrian (3D Detection) | 47.30 % | 39.41 % | 36.96 %
Cyclist (3D Detection) | 77.33 % | 61.99 % | 55.51 %
 
can not reach the PV-RCNN: Point-Voxel Feature Set Abstraction for 3D Object Detection, 
which the result is 
Car (3D Detection) | 90.25 % | 81.43 % | 76.82 %
Pedestrian (3D Detection) | 52.17 % | 43.29 % | 40.29 %
Cyclist (3D Detection) | 78.60 % | 63.77 % | 57.65 %

The gap is clear, is someone Can someone explain that

jihanyang, 2022-10-12:
Our provided pretrained model in model_zoo should match the performance in the table, otherwise, it might be the environment error or data error.

JianyyuWang, 2022-10-13:
> Our provided pretrained model in model_zoo should match the performance in the table, otherwise, it might be the environment error or data error.

hi，I obtained the results in the test set according to the pv_rcnn pre-training file and configuration file provided by OpenPCDet, and uploaded them to the official website of kitti. The results were only output after the kitti test was passed. After being confirmed by the official website, the result does decrease. Can you provide the prediction result of the official website?

my profile is ：
CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']

DATA_CONFIG:
    _BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset_test.yaml
    DATA_AUGMENTOR:
        DISABLE_AUG_LIST: ['placeholder']
        AUG_CONFIG_LIST:
            - NAME: gt_sampling
              USE_ROAD_PLANE: True
              DB_INFO_PATH:
                  - kitti_dbinfos_train.pkl
              PREPARE: {
                 filter_by_min_points: ['Car:5', 'Pedestrian:5', 'Cyclist:5'],
                 filter_by_difficulty: [-1],
              }

              SAMPLE_GROUPS: ['Car:15','Pedestrian:10', 'Cyclist:10']
              NUM_POINT_FEATURES: 4
              DATABASE_WITH_FAKELIDAR: False
              REMOVE_EXTRA_WIDTH: [0.0, 0.0, 0.0]
              LIMIT_WHOLE_SCENE: False

            - NAME: random_world_flip
              ALONG_AXIS_LIST: ['x']

            - NAME: random_world_rotation
              WORLD_ROT_ANGLE: [-0.78539816, 0.78539816]

            - NAME: random_world_scaling
              WORLD_SCALE_RANGE: [0.95, 1.05]

MODEL:
    NAME: PVRCNN

    VFE:
        NAME: MeanVFE

    BACKBONE_3D:
        NAME: VoxelBackBone8x

    MAP_TO_BEV:
        NAME: HeightCompression
        NUM_BEV_FEATURES: 256

    BACKBONE_2D:
        NAME: BaseBEVBackbone

        LAYER_NUMS: [5, 5]
        LAYER_STRIDES: [1, 2]
        NUM_FILTERS: [128, 256]
        UPSAMPLE_STRIDES: [1, 2]
        NUM_UPSAMPLE_FILTERS: [256, 256]

    DENSE_HEAD:
        NAME: AnchorHeadSingle
        CLASS_AGNOSTIC: False

        USE_DIRECTION_CLASSIFIER: True
        DIR_OFFSET: 0.78539
        DIR_LIMIT_OFFSET: 0.0
        NUM_DIR_BINS: 2

        ANCHOR_GENERATOR_CONFIG: [
            {
                'class_name': 'Car',
[... 161 lines cut ...]
            NMS_POST_MAXSIZE: 500

OPTIMIZATION:
    BATCH_SIZE_PER_GPU: 2
    NUM_EPOCHS: 80

    OPTIMIZER: adam_onecycle
    LR: 0.01
    WEIGHT_DECAY: 0.01
    MOMENTUM: 0.9

    MOMS: [0.95, 0.85]
    PCT_START: 0.4
    DIV_FACTOR: 10
    DECAY_STEP_LIST: [35, 45]
    LR_DECAY: 0.1
    LR_CLIP: 0.0000001

    LR_WARMUP: False
    WARMUP_EPOCH: 1

    GRAD_NORM_CLIP: 10

the web result is :
[image]

JianyyuWang, 2022-10-13:
does any param need to be modified?

JianyyuWang, 2022-10-13:
my kitti test config is:
DATASET: 'KittiDataset'
DATA_PATH: '../data/kitti'

POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]

DATA_SPLIT: {
    'train': train,
    'test': test
}

INFO_PATH: {
    'train': [kitti_infos_train.pkl],
    'test': [kitti_infos_test.pkl],
}

GET_ITEM_LIST: ["points"]
FOV_POINTS_ONLY: True

DATA_AUGMENTOR:
    DISABLE_AUG_LIST: ['placeholder']
    AUG_CONFIG_LIST:
        - NAME: gt_sampling
          USE_ROAD_PLANE: True
          DB_INFO_PATH:
              - kitti_dbinfos_train.pkl
          PREPARE: {
             filter_by_min_points: ['Car:5', 'Pedestrian:5', 'Cyclist:5'],
             filter_by_difficulty: [-1],
          }

          SAMPLE_GROUPS: ['Car:20','Pedestrian:15', 'Cyclist:15']
          NUM_POINT_FEATURES: 4
          DATABASE_WITH_FAKELIDAR: False
          REMOVE_EXTRA_WIDTH: [0.0, 0.0, 0.0]
          LIMIT_WHOLE_SCENE: True

        - NAME: random_world_flip
          ALONG_AXIS_LIST: ['x']

        - NAME: random_world_rotation
          WORLD_ROT_ANGLE: [-0.78539816, 0.78539816]

        - NAME: random_world_scaling
          WORLD_SCALE_RANGE: [0.95, 1.05]

POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z', 'intensity'],
    src_feature_list: ['x', 'y', 'z', 'intensity'],
}

DATA_PROCESSOR:
    - NAME: mask_points_and_boxes_outside_range
      REMOVE_OUTSIDE_BOXES: True

    - NAME: shuffle_points
      SHUFFLE_ENABLED: {
        'train': True,
        'test': False
      }

    - NAME: transform_points_to_voxels
      VOXEL_SIZE: [0.05, 0.05, 0.1]
      MAX_POINTS_PER_VOXEL: 5
      MAX_NUMBER_OF_VOXELS: {
        'train': 16000,
        'test': 40000
      }

sshaoshuai, 2022-10-15:
The provided pretrained model is utilized to evaluate and verify the performance on the KITTI validation set, since the KITTI test set performance is not stable. 
Also, as mentioned in the paper of PV-RCNN, the test set results are achieved by training with a customized 80% training set, instead of the 50% training set (3721 samples) as the provided pretrained model.

JianyyuWang, 2022-10-23:
> The provided pretrained model is utilized to evaluate and verify the performance on the KITTI validation set, since the KITTI test set performance is not stable. Also, as mentioned in the paper of [...]

hi，could you please provide sustomized 80% training set profile information？thanks

closed 2022-12-08 by github-actions[bot]