open-mmlab/OpenPCDet #1146: Evaluating the Waymo dataset is slow

yanghtr, 2022-10-12:
I use 8 TITAN V to train centerpoint on Waymo but the AP calculation is very slow (about 10 hours), the log message is below
```
2022-10-09 03:18:44,060   INFO  **********************Start evaluation waymo_models/centerpoint(default)**********************
2022-10-09 03:18:44,064   INFO  Loading Waymo dataset
2022-10-09 03:19:11,705   INFO  Total skipped info 0
2022-10-09 03:19:11,706   INFO  Total samples for Waymo dataset: 39987
2022-10-09 03:19:11,728   INFO  ==> Loading parameters from checkpoint /***/checkpoint_epoch_30.pth to CPU
2022-10-09 03:19:11,803   INFO  ==> Checkpoint trained from version: pcdet+0.5.2+7e8bbe2+pyb345b08
2022-10-09 03:19:11,864   INFO  ==> Done (loaded 279/279)
2022-10-09 03:19:11,869   INFO  *************** EPOCH 30 EVALUATION *****************
2022-10-09 03:37:01,190   INFO  *************** Performance of EPOCH 30 *****************
2022-10-09 03:37:01,190   INFO  Generate label finished(sec_per_example: 0.0267 second).
2022-10-09 03:37:01,190   INFO  recall_roi_0.3: 0.000000
2022-10-09 03:37:01,191   INFO  recall_rcnn_0.3: 0.862386                                                                                                                                                                            2022-10-09 03:37:01,191   INFO  recall_roi_0.5: 0.000000
2022-10-09 03:37:01,191   INFO  recall_rcnn_0.5: 0.795258                                                                                                                                                                            2022-10-09 03:37:01,191   INFO  recall_roi_0.7: 0.000000
2022-10-09 03:37:01,191   INFO  recall_rcnn_0.7: 0.521872                                                                                                                                                                            2022-10-09 03:37:01,219   INFO  Average predicted number of objects(39987 samples): 212.770
2022-10-09 13:47:24,393   INFO
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.7258
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.7204
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APL: 0.7258
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.6446
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.6397
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APL: 0.6446
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.7357
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.6724
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APL: 0.7357
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.6565                                                                                                                                                                                       OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.5986
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APL: 0.6565
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/AP: 0.0000                                                                                                                                                                                             OBJECT_TYPE_TYPE_SIGN_LEVEL_1/APH: 0.0000
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/APL: 0.0000                                                                                                                                                                                            OBJECT_TYPE_TYPE_SIGN_LEVEL_2/AP: 0.0000
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/APH: 0.0000
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/APL: 0.0000
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.7042
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.6926
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APL: 0.7042
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.6783
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.6671
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APL: 0.6783

2022-10-09 13:47:24,415   INFO  Result is save to /***/OpenPCDet/output/waymo_models/centerpoint/default/eval/eval_with_train/epoch_30/val
2022-10-09 13:47:24,415   INFO  ****************Evaluation done.*****************                                                                                                                                                    2022-10-09 13:47:24,567   INFO  Epoch 30 has been evaluated
2022-10-09 13:47:54,599   INFO  **********************End evaluation waymo_models/centerpoint(default)**********************
```

Is this speed normal? Are there any ways to accelerate it? In addition, my results are about 0.5% worse than the results in README.md (consistently), is it normal?

xiaoxin83121, 2022-10-13:
Try install tensorflow-gpu instead, that costs about 30 minutes to evaluate AP.

closed 2022-11-07 by yanghtr

YuRui-Learning, 2024-01-11:
hello, when i run evaluation, i get some bug like this one , could u give me some help 
Traceback (most recent call last):
  File "eval.py", line 207, in <module>
    main()
  File "eval.py", line 203, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "eval.py", line 65, in eval_single_ckpt
    eval_result.eval_one_epoch(
  File "/root/MSF/tools/eval_utils/eval_result.py", line 62, in eval_one_epoch
    result_str, result_dict = dataset.evaluation(
  File "/root/MSF/tools/../pcdet/datasets/waymo/waymo_dataset.py", line 476, in evaluation
    ap_result_str, ap_dict = waymo_eval(eval_det_annos, eval_gt_annos)
  File "/root/MSF/tools/../pcdet/datasets/waymo/waymo_dataset.py", line 459, in waymo_eval
    ap_dict = eval.waymo_evaluation(
  File "/root/MSF/tools/../pcdet/datasets/waymo/waymo_eval.py", line 211, in waymo_evaluation
    metrics = self.build_graph(graph)
  File "/root/MSF/tools/../pcdet/datasets/waymo/waymo_eval.py", line 127, in build_graph
    metrics = detection_metrics.get_detection_metric_ops(
  File "/root/.local/lib/python3.8/site-packages/waymo_open_dataset/metrics/python/detection_metrics.py", line 134, in get_detection_metric_ops
    ground_truth_speed = tf.zeros((num_gt_boxes, 2), tf.float32)
  File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/util/dispatch.py", line 206, in wrapper
    return target(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/ops/array_ops.py", line 2912, in wrapped
    tensor = fun(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/ops/array_ops.py", line 2962, in zeros
    output = _constant_if_small(zero, shape, dtype, name)
  File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/ops/array_ops.py", line 2897, in _constant_if_small
    if np.prod(shape) < 1000:
  File "<__array_function__ internals>", line 200, in prod
  File "/usr/local/lib/python3.8/dist-packages/numpy/core/fromnumeric.py", line 3076, in prod
    return _wrapreduction(a, np.multiply, 'prod', axis, dtype, out,
  File "/usr/local/lib/python3.8/dist-packages/numpy/core/fromnumeric.py", line 86, in _wrapreduction
    return ufunc.reduce(obj, axis, dtype, out, **passkwargs)
  File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/framework/ops.py", line 867, in __array__
    raise NotImplementedError(
NotImplementedError: Cannot convert a symbolic Tensor (strided_slice:0) to a numpy array. This error may indicate that you're trying to pass a Tensor to a NumPy call, which is not supported
eval:   0%|          | 0/4931 [02:31<?, ?it/s]