open-mmlab/OpenPCDet #1147: MPPNet assumes the input point features 6 dim

kan-bayashi, 2022-10-12:
Hi, Thank you for developing a great toolkit!
MPPNet performance seems very attractive and I want to train MPPNet using custom dataset.
At first, I try to train MPPnet with waymo dataset using different point features to match with custom dataset.

I modified the following line:
https://github.com/open-mmlab/OpenPCDet/blob/f221374a5cb9398fd089aa7194732c808c700355/tools/cfgs/waymo_models/mppnet_4frames.yaml#L47-L51
to
```yaml
    POINT_FEATURE_ENCODING: {
        encoding_type: absolute_coordinates_encoding,
        used_feature_list: ['x', 'y', 'z', 'intensity', 'time'],
        src_feature_list: ['x', 'y', 'z', 'intensity', 'elongation', 'time'],
    }
```
But I got the following errors:
```
2022-10-12 19:30:12,575   INFO  **********************Start logging**********************
2022-10-12 19:30:12,575   INFO  CUDA_VISIBLE_DEVICES=2
2022-10-12 19:30:12,575   INFO  cfg_file         cfgs/waymo_models/mppnet_4frames.v2.yaml
2022-10-12 19:30:12,575   INFO  batch_size       2
2022-10-12 19:30:12,575   INFO  epochs           6
2022-10-12 19:30:12,575   INFO  workers          4
2022-10-12 19:30:12,576   INFO  extra_tag        default
2022-10-12 19:30:12,576   INFO  ckpt             None
2022-10-12 19:30:12,576   INFO  pretrained_model None
2022-10-12 19:30:12,576   INFO  launcher         none
2022-10-12 19:30:12,576   INFO  tcp_port         18888
2022-10-12 19:30:12,576   INFO  sync_bn          False
2022-10-12 19:30:12,576   INFO  fix_random_seed  False
2022-10-12 19:30:12,576   INFO  ckpt_save_interval 1
2022-10-12 19:30:12,577   INFO  local_rank       0
2022-10-12 19:30:12,577   INFO  max_ckpt_save_num 30
2022-10-12 19:30:12,577   INFO  merge_all_iters_to_one_epoch False
2022-10-12 19:30:12,577   INFO  set_cfgs         ['DATA_CONFIG.ROI_BOXES_PATH.train', '/work/dl-user/OpenPCDet/examples/waymo_mppnet/exp/centerpoint_4frames.v2/default/eval/epoch_10/train/default/result.pkl', 'DATA_CONFIG.ROI_BOXES_PATH.test', '/work/dl-user/OpenPCDet/examples/waymo_mppnet/exp/centerpoint_4frames.v2/default/eval/epoch_10/val/default/result.pkl']
2022-10-12 19:30:12,577   INFO  max_waiting_mins 0
2022-10-12 19:30:12,577   INFO  start_epoch      0
2022-10-12 19:30:12,577   INFO  num_epochs_to_eval 0
2022-10-12 19:30:12,577   INFO  save_to_file     False
2022-10-12 19:30:12,577   INFO  use_tqdm_to_record False
2022-10-12 19:30:12,577   INFO  logger_iter_interval 50
2022-10-12 19:30:12,577   INFO  ckpt_save_time_interval 18000
2022-10-12 19:30:12,577   INFO  wo_gpu_stat      False
2022-10-12 19:30:12,577   INFO  cfg.ROOT_DIR: /work/dl-user/OpenPCDet
2022-10-12 19:30:12,578   INFO  cfg.LOCAL_RANK: 0
2022-10-12 19:30:12,578   INFO  cfg.CLASS_NAMES: ['Vehicle', 'Pedestrian', 'Cyclist']
2022-10-12 19:30:12,578   INFO
cfg.DATA_CONFIG = edict()
2022-10-12 19:30:12,578   INFO  cfg.DATA_CONFIG.DATASET: WaymoDataset
2022-10-12 19:30:12,578   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/waymo
2022-10-12 19:30:12,578   INFO  cfg.DATA_CONFIG.PROCESSED_DATA_TAG: waymo_processed_data_v0_5_0
2022-10-12 19:30:12,578   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [-75.2, -75.2, -2, 75.2, 75.2, 4]
2022-10-12 19:30:12,578   INFO
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2022-10-12 19:30:12,578   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2022-10-12 19:30:12,578   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2022-10-12 19:30:12,578   INFO
cfg.DATA_CONFIG.SAMPLED_INTERVAL = edict()
2022-10-12 19:30:12,578   INFO  cfg.DATA_CONFIG.SAMPLED_INTERVAL.train: 1
2022-10-12 19:30:12,578   INFO  cfg.DATA_CONFIG.SAMPLED_INTERVAL.test: 1
2022-10-12 19:30:12,578   INFO  cfg.DATA_CONFIG.FILTER_EMPTY_BOXES_FOR_TRAIN: True
2022-10-12 19:30:12,579   INFO  cfg.DATA_CONFIG.DISABLE_NLZ_FLAG_ON_POINTS: True
2022-10-12 19:30:12,579   INFO  cfg.DATA_CONFIG.USE_SHARED_MEMORY: False
2022-10-12 19:30:12,579   INFO  cfg.DATA_CONFIG.SHARED_MEMORY_FILE_LIMIT: 35000
2022-10-12 19:30:12,579   INFO
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2022-10-12 19:30:12,579   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2022-10-12 19:30:12,579   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x', 'y']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2022-10-12 19:30:12,579   INFO
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2022-10-12 19:30:12,579   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
[... 456 lines cut ...]
Traceback (most recent call last):                                                                        | 0/79041 [00:00<?, ?it/s]
  File "train.py", line 221, in <module>
    main()
  File "train.py", line 190, in main
    show_gpu_stat=not args.wo_gpu_stat
  File "/work/dl-user/OpenPCDet/tools/train_utils/train_utils.py", line 162, in train_model
    show_gpu_stat=show_gpu_stat
  File "/work/dl-user/OpenPCDet/tools/train_utils/train_utils.py", line 51, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "../pcdet/models/__init__.py", line 42, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/work/dl-user/OpenPCDet/venv/.venv/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "../pcdet/models/detectors/mppnet.py", line 20, in forward
    batch_dict = cur_module(batch_dict)
  File "/work/dl-user/OpenPCDet/venv/.venv/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "../pcdet/models/roi_heads/mppnet_head.py", line 704, in forward
    src = self.crop_previous_frame_points(src, batch_size,trajectory_rois, num_rois,valid_length,batch_dict)
  File "../pcdet/models/roi_heads/mppnet_head.py", line 545, in crop_previous_frame_points
    src[bs_idx, roi_box_idx, self.num_lidar_points*idx:self.num_lidar_points*(idx+1), :] = cur_roi_points_sample
RuntimeError: The expanded size of the tensor (5) must match the existing size (4) at non-singleton dimension 1.  Target sizes: [128, 5].  Tensor sizes: [128, 4]
```
Maybe some parts of MPPnet is hard coded for the 6 dimensional inputs.
Could you give me some hints to solve this issue?

Cedarch, 2022-10-14:
Hi, you can change the dim of  points here https://github.com/open-mmlab/OpenPCDet/blob/f221374a5cb9398fd089aa7194732c808c700355/pcdet/models/roi_heads/mppnet_head.py#L700

kan-bayashi, 2022-10-14:
I solved this issue by changing the following parts to refer the point dimension.
https://github.com/open-mmlab/OpenPCDet/blob/f221374a5cb9398fd089aa7194732c808c700355/pcdet/models/roi_heads/mppnet_head.py#L320
https://github.com/open-mmlab/OpenPCDet/blob/f221374a5cb9398fd089aa7194732c808c700355/pcdet/models/roi_heads/mppnet_head.py#L486
https://github.com/open-mmlab/OpenPCDet/blob/f221374a5cb9398fd089aa7194732c808c700355/pcdet/models/roi_heads/mppnet_head.py#L534
https://github.com/open-mmlab/OpenPCDet/blob/f221374a5cb9398fd089aa7194732c808c700355/pcdet/models/roi_heads/mppnet_head.py#L538
https://github.com/open-mmlab/OpenPCDet/blob/f221374a5cb9398fd089aa7194732c808c700355/pcdet/models/roi_heads/mppnet_head.py#L700

Thank you for your suggestion.

closed 2022-10-14 by kan-bayashi

geoseb94, 2023-03-11:
Hi [kan-bayashi](https://github.com/kan-bayashi), 
I am facing the same issue. I am trying to run MPPNet on a custom dataset, which doesn't have 'elongation' feature. My config is as below.

    POINT_FEATURE_ENCODING: {
        encoding_type: absolute_coordinates_encoding,
        used_feature_list: ['x', 'y', 'z', 'intensity', 'time'],
        src_feature_list: ['x', 'y', 'z', 'intensity', 'time'],
    }
    
Thanks for pointing to the 5 lines in the mppnet_head.py, which requires further modification. Could you please let me know the exact modification you made to run the network?

Tottowich, 2023-07-16:
@geoseb94  Hi! I am also looking into using a MPPNet model on a custom, KITTI format, dataset. Have you made any progress on this? I also have ['x', 'y', 'z', 'intensity', 'time'].