open-mmlab/OpenPCDet #1150: There is some errors about 'create_kitti_infos'.

oneway3124, 2022-10-19:
When I execute this following command:
`python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml`

There are some errors as following,
`train sample_idx: 007476
train sample_idx: 007479
Traceback (most recent call last):
  File "/home/lijiadi/anaconda3/envs/OpenPCDet-ww/lib/python3.8/runpy.py", line 192, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/lijiadi/anaconda3/envs/OpenPCDet-ww/lib/python3.8/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/lijiadi/wangwei/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 479, in <module>
    create_kitti_infos(
  File "/home/lijiadi/wangwei/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 443, in create_kitti_infos
    kitti_infos_train = dataset.get_infos(num_workers=workers, has_label=True, count_inside_pts=True)
  File "/home/lijiadi/wangwei/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 222, in get_infos
    return list(infos)
  File "/home/lijiadi/anaconda3/envs/OpenPCDet-ww/lib/python3.8/concurrent/futures/_base.py", line 611, in result_iterator
    yield fs.pop().result()
  File "/home/lijiadi/anaconda3/envs/OpenPCDet-ww/lib/python3.8/concurrent/futures/_base.py", line 432, in result
    return self.__get_result()
  File "/home/lijiadi/anaconda3/envs/OpenPCDet-ww/lib/python3.8/concurrent/futures/_base.py", line 388, in __get_result
    raise self._exception
  File "/home/lijiadi/anaconda3/envs/OpenPCDet-ww/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/lijiadi/wangwei/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 203, in process_single_scene
    points = self.get_lidar(sample_idx)
  File "/home/lijiadi/wangwei/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 66, in get_lidar
    return np.fromfile(str(lidar_file), dtype=np.float32).reshape(-1, 4)
ValueError: cannot reshape array of size 86881 into shape (4)
(OpenPCDet-ww) lijiadi@lijiadi-Z490-UD:~/wangwei/OpenPCDet$ `

Can you help me about this issue?

xuchangjia, 2022-11-01:
did your kitti training dataset is 7481?  "train sample_idx: 007476 train sample_idx: 007479",maybe you should check your kitii dataset.

oneway3124, 2022-11-08:
I have checked these problem that there are 7481 samples for our dataset. I donot know why the log like this status?

closed 2022-12-23 by github-actions[bot]