open-mmlab/OpenPCDet #1151: There are some errors when using the train.py

oneway3124, 2022-10-19:
When I use the following command,
`python train.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml `

The error informations are as following,
`Traceback (most recent call last):
  File "train.py", line 221, in <module>
    main()
  File "train.py", line 111, in main
    train_set, train_loader, train_sampler = build_dataloader(
  File "../pcdet/datasets/__init__.py", line 53, in build_dataloader
    dataset = __all__[dataset_cfg.DATASET](
  File "../pcdet/datasets/kitti/kitti_dataset.py", line 23, in __init__
    super().__init__(
  File "../pcdet/datasets/dataset.py", line 30, in __init__
    self.data_augmentor = DataAugmentor(
  File "../pcdet/datasets/augmentor/data_augmentor.py", line 23, in __init__
    cur_augmentor = getattr(self, cur_cfg.NAME)(config=cur_cfg)
  File "../pcdet/datasets/augmentor/data_augmentor.py", line 27, in gt_sampling
    db_sampler = database_sampler.DataBaseSampler(
  File "../pcdet/datasets/augmentor/database_sampler.py", line 35, in __init__
    sampler_cfg.DB_INFO_PATH[0] = sampler_cfg.BACKUP_DB_INFO['DB_INFO_PATH']
AttributeError: 'EasyDict' object has no attribute 'BACKUP_DB_INFO'
(OpenPCDet-ww) lijiadi@lijiadi-Z490-UD:~/wangwei/OpenPCDet/tools$ `

Can you help me with this error?

Chris-ctrl123, 2022-10-20:
I have the same error

Chris-ctrl123, 2022-10-20:
This error disappeared when I used ‘python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml’ to generate a dictionary of KITTI data

oneway3124, 2022-10-20:
https://github.com/open-mmlab/OpenPCDet/issues/1150  Please see this issue.

xuchangjia, 2022-10-31:
I have same probelm as you,i just cd tools/ then run python train.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml.finally i fix it .

oneway3124, 2022-10-31:
How to fix it? I cannot understand it.

xuchangjia, 2022-10-31:
> 
应该是路径问题，我切到tools路径下运行不报错，运行python tools/train.py ××× 就报这个错。

oneway3124, 2022-10-31:
https://github.com/open-mmlab/OpenPCDet/issues/1150   @xuchangjia Did you have a smiliar issue?

SkywalkerHe, 2022-11-03:
I have same question :AttributeError:` 'EasyDict' object has no attribute 'BACKUP_DB_INFO'`,when I run
`$  bash scripts/dist_train.sh ${NUM_GPUS} --cfg_file cfgs/waymo_models/centerpoint_4frames.yaml`

Anutlet, 2022-12-05:
> > 
> 
> 应该是路径问题，我切到tools路径下运行不报错，运行python tools/train.py ××× 就报这个错。

我切到tools路径下，也还是报错，我跑的nuscense v1.0_mini

closed 2023-01-21 by github-actions[bot]