open-mmlab/OpenPCDet #1152: Unresolved reference 'pointnet2_stack_cuda'

910399614, 2022-10-19:
I try to debug Voxel-RCNN and I find a unresolved reference 'pointnet2_stack_cuda' in the voxel_query_utils.py. But the method of this reference that 'voxel_query_wrapper()' was executed successfully.

[image]
[image]

910399614, 2022-10-20:
> 

It’s difficult to say, but all commands I have run are following official steps

jihanyang, 2022-10-20:
I mean whether you try to run any models and meet with errors?

910399614, 2022-10-20:
> I mean whether you try to run any models and meet with errors?

I have not met any errors, and the model 'Voxel_RCNN' was run successfully, but I want to know the details of method 'voxel_query_wrapper()'.

jihanyang, 2022-10-20:
https://github.com/open-mmlab/OpenPCDet/blob/f221374a5cb9398fd089aa7194732c808c700355/pcdet/ops/pointnet2/pointnet2_stack/src/pointnet2_api.cpp#L14

closed 2022-10-20 by jihanyang

910399614, 2022-10-20:
Thanks for your help
 

姜仪东
***@***.***

 
https://github.com/open-mmlab/OpenPCDet/blob/f221374a5cb9398fd089aa7194732c808c700355/pcdet/ops/pointnet2/pointnet2_stack/src/pointnet2_api.cpp#L14