open-mmlab/OpenPCDet #1153: No module named 'pcdet.version'

bigsheep2012, 2022-10-20:
Hi OpenPCDet team @jihanyang @Cedarch ,
Error occured when trying to create waymo infos with command : 
python -m pcdet.datasets.waymo.waymo_dataset --func create_waymo_infos    --cfg_file tools/cfgs/dataset_configs/waymo_dataset_multiframe.yaml

it seems the file(root/pcdet/version.py) disappeared in newly released v0.6.0 repo.
Could you guys plz fix it? or I can just use previous version.py?

jihanyang, 2022-10-20:
https://github.com/open-mmlab/OpenPCDet/blob/f221374a5cb9398fd089aa7194732c808c700355/setup.py#L32
It should still exist. Have you run `python setup.py develop`?

closed 2022-10-20 by bigsheep2012

wangyakunn, 2023-03-10:
I can run sucessfully(import pcdet) in /OpenPCDet, but failed in /OpenPCDet/tools.I don't know why.