open-mmlab/OpenPCDet #1154: How to run an inference on a data set with a model pretrained pointrcnn for example

EstevanCandido, 2022-10-20:
Thank you for your great work in making the OpenPCDet project available for multi-object detection.

I followed Get Started to do a test on kitti's dataset but the command "python test.py --cfg_file cfgs/kitti_models/pointrcnn.yaml --ckpt /home/lume/OpenPCDet/stev_models/pointrcnn_7870.pth" didn't work for inference on my dataset downloaded from the official kitti.

OBS1: I followed the organization of folders as requested by the Get Started documentation.

I would just like to make an inference, can you help me?

I need set some path of my data set for inference?

[image]

[image]

EstevanCandido, 2022-10-20:
When i run the demo works fine. But i need run in data set of kitti official.

[image]

jihanyang, 2022-10-21:
It seems that you can already obtain the evaluation results. You want to make visualization on kitti validation set?

EstevanCandido, 2022-10-21:
@jihanyang Exactly, I want to visualize the detection of my kitti dataset (not just one image, but all of them, all at once). Could you help me?

EstevanCandido, 2022-10-24:
Another question, when I run test.py am I training and doing the evaluate inferring my dataset for 3D detection? My objective is to evaluate the model in my dataset and in the output I would like the visualization to be a video of my trajectory with the car, for example. Could you help me?

jihanyang, 2022-10-25:
Sorry for late reply. You can add such visualization command  https://github.com/open-mmlab/OpenPCDet/blob/f221374a5cb9398fd089aa7194732c808c700355/tools/demo.py#L100-L103 to https://github.com/open-mmlab/OpenPCDet/blob/f221374a5cb9398fd089aa7194732c808c700355/tools/eval_utils/eval_utils.py#L66-L67

jihanyang, 2022-10-25:
For the video, I don't have any experience on it.

EstevanCandido, 2022-10-25:
@jihanyang I did add the lines of demo.py

https://github.com/open-mmlab/OpenPCDet/blob/f221374a5cb9398fd089aa7194732c808c700355/tools/demo.py#L6-L7 

 under of line 9
 
 https://github.com/open-mmlab/OpenPCDet/blob/f221374a5cb9398fd089aa7194732c808c700355/tools/eval_utils/eval_utils.py#L9-L11

and did add

https://github.com/open-mmlab/OpenPCDet/blob/f221374a5cb9398fd089aa7194732c808c700355/tools/demo.py#L100-L103

under of line 65, replacing **data_dict** to **batch_dict** , because if not an error called "data dict is not defined"

https://github.com/open-mmlab/OpenPCDet/blob/f221374a5cb9398fd089aa7194732c808c700355/tools/eval_utils/eval_utils.py#L65-L67

and the output is printed from image 
[image]

Note that the eval stays at zero even if I press ESC or Q to skip image and it returns eval: 0%

jihanyang, 2022-11-26:
Hello @EstevanCandido, you code is correct. I also run visualization as you did. In my experience, the evaluation should continue after you close the open3d visualization box.
If you comment the added visualization codes, will the program run normal?

closed 2023-01-11 by github-actions[bot]