open-mmlab/OpenPCDet #1156: Trouble evaluating the model with custom dataset

5anyamAgrawal, 2022-10-23:
@jihanyang @MartinHahner  I'm trying to train the model using this repo's newly uploaded custom dataset feature. My goal is to predict sites(pieces of land) using the lidar point cloud data we have. I have followed the steps from the custom dataset tutorials and converted the points clouds to .npy files and adjusted the config file accordingly. However, when I train the model my loss function starts with about 20-30 in the first iteration and then goes to 5-10 and stays within the range for the rest of the epochs which I believe is a little suspicious as it might not be training at all. A quick look the data gave the result that my largest possible anchor box is [58.26, 40.2, 24.02] with a heading angle of range [-3.14, 3.14]. I have adjusted the config file for my model accordingly as well. Additionally, I wasn't able to understand how to modify the point cloud data  range and voxel sizes to meet the rules describe in the custom dataset tutorials. I tried the point cloud range [-40, -40, 0, 40, 40, 40] , but it won't work and also, I'm confused on how the voxel sizes are working. Lastly, I would appreciate if you could explain why my recall is 0 and what does bboxes, bev, 3d, and aos mean ?Do I even need it because I'm only using sites for my data?

**Following is my custom_dataset.yaml file:**
`DATASET: 'CustomDataset'
DATA_PATH: '/work/archcomplab/sagrawal/latest_openpcdet/OpenPCDet/data/custom'

POINT_CLOUD_RANGE: [-40, -40, -3, 70.4, 40, 1]

CLASS_NAMES: ['Site'] 

MAP_CLASS_TO_KITTI: {
    'Site': 'Car'
}

DATA_SPLIT: {
    'train': train,
    'test': val
}

INFO_PATH: {
    'train': [custom_infos_train.pkl],
    'test': [custom_infos_val.pkl],
}

POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z', 'intensity'],
    src_feature_list: ['x', 'y', 'z', 'intensity'],
}

DATA_AUGMENTOR:
    DISABLE_AUG_LIST: ['placeholder']
    AUG_CONFIG_LIST:
        # - NAME: gt_sampling
        #   USE_ROAD_PLANE: False
        #   DB_INFO_PATH:
        #       - custom_dbinfos_train.pkl
        #   PREPARE: {
        #      filter_by_min_points: ['Site:15']
        #   }

        #   SAMPLE_GROUPS: ['Site:50']
        #   NUM_POINT_FEATURES: 4
        #   DATABASE_WITH_FAKELIDAR: False
        #   REMOVE_EXTRA_WIDTH: [0.0, 0.0, 0.0]
        #   LIMIT_WHOLE_SCENE: True

        - NAME: random_world_flip
          ALONG_AXIS_LIST: ['x', 'y']

        - NAME: random_world_rotation
          WORLD_ROT_ANGLE: [-0.78539816, 0.78539816]

        - NAME: random_world_scaling
          WORLD_SCALE_RANGE: [0.95, 1.05]

DATA_PROCESSOR:
    - NAME: mask_points_and_boxes_outside_range
      REMOVE_OUTSIDE_BOXES: True

    - NAME: shuffle_points
      SHUFFLE_ENABLED: {
        'train': True,
        'test': False
      }

    - NAME: transform_points_to_voxels
      VOXEL_SIZE: [0.05, 0.05, 0.1]
      MAX_POINTS_PER_VOXEL: 5
      MAX_NUMBER_OF_VOXELS: {
        'train': 16000,
        'test': 40000
      }`

**Following is my second.yaml:**

`CLASS_NAMES: ['Site']

DATA_CONFIG:
    _BASE_CONFIG_: cfgs/dataset_configs/custom_dataset.yaml

MODEL:
    NAME: SECONDNet

    VFE:
        NAME: MeanVFE

    BACKBONE_3D:
        NAME: VoxelBackBone8x

    MAP_TO_BEV:
        NAME: HeightCompression
        NUM_BEV_FEATURES: 256

    BACKBONE_2D:
        NAME: BaseBEVBackbone

        LAYER_NUMS: [5, 5]
        LAYER_STRIDES: [1, 2]
        NUM_FILTERS: [128, 256]
        UPSAMPLE_STRIDES: [1, 2]
        NUM_UPSAMPLE_FILTERS: [256, 256]

    DENSE_HEAD:
        NAME: AnchorHeadSingle
        CLASS_AGNOSTIC: False

        USE_DIRECTION_CLASSIFIER: False
        DIR_OFFSET: 0.78539
        DIR_LIMIT_OFFSET: 0.0
        NUM_DIR_BINS: 2

        ANCHOR_GENERATOR_CONFIG: [
            {
                'class_name': 'Site',
                'anchor_sizes': [[58.26, 40.2 , 24.02]],
                'anchor_rotations': [-3.14, 3.14],
                'anchor_bottom_heights': [0],
                'align_center': False,
                'feature_map_stride': 8,
                'matched_threshold': 0.5,
                'unmatched_threshold': 0.35
            }
        ]

        TARGET_ASSIGNER_CONFIG:
            NAME: AxisAlignedTargetAssigner
            POS_FRACTION: -1.0
            SAMPLE_SIZE: 512
            NORM_BY_NUM_EXAMPLES: False
            MATCH_HEIGHT: False
            BOX_CODER: ResidualCoder

        LOSS_CONFIG:
            LOSS_WEIGHTS: {
                'cls_weight': 1.0,
                'loc_weight': 2.0,
                'dir_weight': 0.2,
                'code_weights': [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
            }

    POST_PROCESSING:
        RECALL_THRESH_LIST: [0.3, 0.5, 0.7]
        SCORE_THRESH: 0.1
        OUTPUT_RAW_SCORE: False

        EVAL_METRIC: kitti

        NMS_CONFIG:
            MULTI_CLASSES_NMS: False
            NMS_TYPE: nms_gpu
            NMS_THRESH: 0.50
            NMS_PRE_MAXSIZE: 4096
            NMS_POST_MAXSIZE: 500

OPTIMIZATION:
    BATCH_SIZE_PER_GPU: 4
    NUM_EPOCHS: 80

    OPTIMIZER: adam_onecycle
    LR: 0.003
    WEIGHT_DECAY: 0.01
    MOMENTUM: 0.9

    MOMS: [0.95, 0.85]
    PCT_START: 0.4
    DIV_FACTOR: 10
    DECAY_STEP_LIST: [35, 45]
    LR_DECAY: 0.1
    LR_CLIP: 0.0000001

    LR_WARMUP: False
    WARMUP_EPOCH: 1

    GRAD_NORM_CLIP: 10`

5anyamAgrawal, 2022-10-23:
**My training looks something like this:**
`epochs:   0%|          | 0/120 [00:00<?, ?it/s]

train:   0%|          | 0/271 [00:00<?, ?it/s][A2022-10-23 03:09:22,423   INFO  epoch: 0/120, acc_iter=1, cur_iter=0/271, batch_size=4, loss=22.747303009033203, lr=0.0002999999999999999
2022-10-23 03:09:45,955   INFO  epoch: 0/120, acc_iter=50, cur_iter=49/271, batch_size=4, loss=8.649927139282227, lr=0.00030009452997724123
2022-10-23 03:10:13,163   INFO  epoch: 0/120, acc_iter=100, cur_iter=99/271, batch_size=4, loss=4.368033409118652, lr=0.00030038586213345514
2022-10-23 03:10:39,582   INFO  epoch: 0/120, acc_iter=150, cur_iter=149/271, batch_size=4, loss=4.477015495300293, lr=0.0003008739933584902
2022-10-23 03:10:42,033   INFO  c2613.crane.hcc.unl.edu  Sun Oct 23 03:10:42 2022  515.65.01
[0] Tesla V100-PCIE-16GB | 56°C,   0 % | 10160 / 16384 MB | sagrawal(9933M)

2022-10-23 03:11:06,952   INFO  epoch: 0/120, acc_iter=200, cur_iter=199/271, batch_size=4, loss=5.329353332519531, lr=0.00030155885247365455
2022-10-23 03:11:33,223   INFO  epoch: 0/120, acc_iter=250, cur_iter=249/271, batch_size=4, loss=7.2046027183532715, lr=0.00030244033961363556
2022-10-23 03:11:41,629   INFO  epoch: 0/120, acc_iter=271, cur_iter=270/271, batch_size=1, loss=4.94480562210083, lr=0.0003028691687696665

                                              [A
epochs:   1%|          | 1/120 [02:44<5:26:28, 164.61s/it]

train:   0%|          | 0/271 [00:00<?, ?it/s][A2022-10-23 03:11:46,587   INFO  epoch: 1/120, acc_iter=272, cur_iter=0/271, batch_size=4, loss=4.262301921844482, lr=0.00030289045362788536
2022-10-23 03:11:58,842   INFO  epoch: 1/120, acc_iter=300, cur_iter=28/271, batch_size=4, loss=5.504721641540527, lr=0.00030351832624106673
2022-10-23 03:12:01,933   INFO  c2613.crane.hcc.unl.edu  Sun Oct 23 03:12:01 2022  515.65.01
[0] Tesla V100-PCIE-16GB | 56°C,   0 % | 10160 / 16384 MB | sagrawal(9933M)

2022-10-23 03:12:27,323   INFO  epoch: 1/120, acc_iter=350, cur_iter=78/271, batch_size=4, loss=6.363627910614014, lr=0.00030479265516526787
2022-10-23 03:12:52,756   INFO  epoch: 1/120, acc_iter=400, cur_iter=128/271, batch_size=4, loss=4.672891616821289, lr=0.00030626314056516604
2022-10-23 03:13:17,689   INFO  epoch: 1/120, acc_iter=450, cur_iter=178/271, batch_size=4, loss=6.176321983337402, lr=0.00030792956801639496
2022-10-23 03:13:19,589   INFO  c2613.crane.hcc.unl.edu  Sun Oct 23 03:13:19 2022  515.65.01
[0] Tesla V100-PCIE-16GB | 56°C,   0 % | 10164 / 16384 MB | sagrawal(9937M)

2022-10-23 03:13:44,299   INFO  epoch: 1/120, acc_iter=500, cur_iter=228/271, batch_size=4, loss=4.097621917724609, lr=0.00030979169452255905
2022-10-23 03:14:03,555   INFO  epoch: 1/120, acc_iter=542, cur_iter=270/271, batch_size=1, loss=8.806389808654785, lr=0.0003115069192024043

                                              [A
epochs:   2%|▏         | 2/120 [05:06<4:57:32, 151.29s/it]

train:   0%|          | 0/271 [00:00<?, ?it/s][A2022-10-23 03:14:08,620   INFO  epoch: 2/120, acc_iter=543, cur_iter=0/271, batch_size=4, loss=4.6472649574279785, lr=0.0003115494371453559
2022-10-23 03:14:11,695   INFO  epoch: 2/120, acc_iter=550, cur_iter=7/271, batch_size=4, loss=4.298891067504883, lr=0.0003118492485506678
2022-10-23 03:14:37,966   INFO  epoch: 2/120, acc_iter=600, cur_iter=57/271, batch_size=4, loss=4.121959209442139, lr=0.0003141019300707316
2022-10-23 03:14:40,987   INFO  c2613.crane.hcc.unl.edu  Sun Oct 23 03:14:40 2022  515.65.01
[0] Tesla V100-PCIE-16GB | 55°C,   0 % | 10164 / 16384 MB | sagrawal(9937M)

2022-10-23 03:15:02,813   INFO  epoch: 2/120, acc_iter=650, cur_iter=107/271, batch_size=4, loss=3.572237253189087, lr=0.0003165494105995094
2022-10-23 03:15:29,501   INFO  epoch: 2/120, acc_iter=700, cur_iter=157/271, batch_size=4, loss=4.063368320465088, lr=0.00031919133324840926
2022-10-23 03:15:55,897   INFO  epoch: 2/120, acc_iter=750, cur_iter=207/271, batch_size=4, loss=4.1347808837890625, lr=0.00032202731277552977
2022-10-23 03:15:59,394   INFO  c2613.crane.hcc.unl.edu  Sun Oct 23 03:15:59 2022  515.65.01
[0] Tesla V100-PCIE-16GB | 55°C,   0 % | 10164 / 16384 MB | sagrawal(9937M)

2022-10-23 03:16:21,787   INFO  epoch: 2/120, acc_iter=800, cur_iter=257/271, batch_size=4, loss=4.845620155334473, lr=0.0003250569356418346
2022-10-23 03:16:26,663   INFO  epoch: 2/120, acc_iter=813, cur_iter=270/271, batch_size=1, loss=3.8342325687408447, lr=0.00032587630249631327

                                              [A
epochs:   2%|▎         | 3/120 [07:29<4:47:44, 147.56s/it]

train:   0%|          | 0/271 [00:00<?, ?it/s][A2022-10-23 03:16:29,809   INFO  epoch: 3/120, acc_iter=814, cur_iter=0/271, batch_size=4, loss=4.802083969116211, lr=0.0003259398714556389
2022-10-23 03:16:50,261   INFO  epoch: 3/120, acc_iter=850, cur_iter=36/271, batch_size=4, loss=4.411231994628906, lr=0.0003282797600714536
2022-10-23 03:17:16,718   INFO  epoch: 3/120, acc_iter=900, cur_iter=86/271, batch_size=4, loss=5.106842994689941, lr=0.000331695316116105
2022-10-23 03:17:18,555   INFO  c2613.crane.hcc.unl.edu  Sun Oct 23 03:17:18 2022  515.65.01
[0] Tesla V100-PCIE-16GB | 55°C,   0 % | 10164 / 16384 MB | sagrawal(9937M)

2022-10-23 03:17:40,469   INFO  epoch: 3/120, acc_iter=950, cur_iter=136/271, batch_size=4, loss=6.504666328430176, lr=0.000335303105723619
2022-10-23 03:18:06,772   INFO  epoch: 3/120, acc_iter=1000, cur_iter=186/271, batch_size=4, loss=4.0181779861450195, lr=0.00033910260281056447
[... cut ...]
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:22.23, 22.23, 22.23
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:35.5524, 35.5524, 35.5524
bev  AP:0.0003, 0.0003, 0.0003
3d   AP:0.0003, 0.0003, 0.0003
aos  AP:17.30, 17.30, 17.30
Car AP@0.70, 0.50, 0.50:
bbox AP:38.8774, 38.8774, 38.8774
bev  AP:0.0106, 0.0106, 0.0106
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:22.23, 22.23, 22.23
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:35.5524, 35.5524, 35.5524
bev  AP:0.0076, 0.0076, 0.0076
3d   AP:0.0076, 0.0076, 0.0076
aos  AP:17.30, 17.30, 17.30`

5anyamAgrawal, 2022-10-23:
Any help would be appreciated!

MartinHahner, 2022-10-23:
Sorry, I don't work with this repository anymore.

closed 2022-12-09 by github-actions[bot]

AJAY31797, 2024-01-19:
Hey @5anyamAgrawal, 

Were you able to figure out what exactly these values in the output mean? I am also trying Secondnet on some site point clouds and getting results in very similar fashion.  But not really able to interpret the results. 

Thanks.